espressif/esp-matter #830: unable to build esp-matter Light example from cleanly cloned repos

MarleenDev, 2024-02-07:
**Describe the bug**
Compile error while trying to build the example.
To me the error looks like a C++ compiler is being used instead of a C compiler when encountering the header file where the error appears. Namespace is a reserved keyword in C++.

**Environment**
- ESP-Matter Commit Id: 886aa35 (latest on main branch as of 7 Feb 2024)
- ESP-IDF Commit Id: 5454d37d49 (latest on master branch as of 7 feb 2024)
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6
- Device Logs (Please attach the log file):
[idf_py_stdout_output_1030498.txt](https://github.com/espressif/esp-matter/files/14191270/idf_py_stdout_output_1030498.txt)
- Host Machine OS: xubuntu 22.04LTS
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
Because I kept running into this error, I decided to do a clean install, and followed all the instructions on the webpages of esp-idf and esp-matter when cloning the repositories and sourcing the export.sh scripts.
Since I managed to get it working in a previous commit, I will try to  find where exactly the error got introduced.

MarleenDev, 2024-02-07:
with esp-idf checked out at v5.3-dev and esp-matter at latest main branch commit, it builds fine

MarleenDev, 2024-02-07:
using git bisect, I find the following commit is the first where the example doesn't build anymore:

commit 92d20f6708ccfe43098e66cab023ba9de2e2bb02 (HEAD, refs/bisect/bad)
Author: SumeetSingh19 <sumeet.singh@espressif.com>
Date:   Thu Jan 25 18:25:40 2024 +0530

    feat(nimble): client presentation and aggregate format descriptor support

diff --git a/components/bt/host/nimble/nimble b/components/bt/host/nimble/nimble
index e6c502f259..0273e0e83a 160000
--- a/components/bt/host/nimble/nimble
+++ b/components/bt/host/nimble/nimble
@@ -1 +1 @@
-Subproject commit e6c502f259812a6d09064fb684e3b56195c80341
+Subproject commit 0273e0e83a05559809c317e41fa4fd88a244620e

dhrishi, 2024-02-07:
@MarleenDev As per the [documentation](https://github.com/espressif/esp-matter?tab=readme-ov-file#supported-esp-idf-and-connectedhomeip-versions), the ESP-IDF version supported is v5.1.2.

On a related note, this issue is fixed internally in ESP-IDF and the fix will be pushed to Github soon.

closed 2024-02-07 by dhrishi