espressif/esp-matter #832: ESP-MATTER Controller: Unable to build with commissioning enabled

nichonaugle, 2024-02-08:
**Describe the bug**
Issue with the ESP-MATTER-BR sketch on build when "Enable matter commissioner" option is enabled.

**Environment**
- ESP-Matter Commit Id: a301a3d
- ESP-IDF Commit Id: v5.1.2
- SoC (eg: ESP32 or ESP32-C3): ESP32S3 & ESP32C6
- Device Logs (Please attach the log file): n/a
- Host Machine OS: MacOS
- Commissioner app and versions if present: n/a
- Commissioner's logs if present: n/a

When compiling the code, the controller example is unable to build and fails with a strange error when "Enable matter commissioner" option is enabled. I could not find very much development documentation or setup instructions which would help me resolve this issue. There are other options such as "Enable Matter over Thread" which have no information on whether they should be enabled alongside the "Enable matter commissioner" option. I have the ESP32S3 connected to a ESP32C6 via UART with OpenThread RCP sketch uploaded to it. 

The controller example compiles and functions correctly when uploaded to ESP32S3 however fails to build whenever "Enable matter commissioner" option is enabled.

Error: [1535/1537] Linking CXX executable controller.elf FAILED: controller.elf

nichonaugle, 2024-02-08:
I have attached the output log below:

[commisioner_error_logs.txt](https://github.com/espressif/esp-matter/files/14215362/commisioner_error_logs.txt)

jakabernat, 2024-02-09:
I am having same problem as well. Re-tested it using the same set up. Hope this gets resolved quickly.

shubhamdp, 2024-02-09:
@nichonaugle  @jakabernat can you specify the build command that you have used? Our CI do build the controller app with thread border router support and it is green. I tried building locally and it worked too.
```
cd examples/controller
idf.py set-target esp32s3 build

# and to build with TRB support you need to build with the `sdkconfig.defaults.otbr` config file
idf.py -D SDKCONFIG_DEFAULTS="sdkconfig.defaults.otbr" set-target esp32s3 build
```

In your build logs, it looks like they are unable to find the references to commissioner related functions from connectedhomeip. Can you check if you have the submodule updated?

Also, if you are doing any additional stuff, let us know.

nichonaugle, 2024-02-09:
I have run the exact command you have shown above in the following order:

cd examples/controller
idf.py -D SDKCONFIG_DEFAULTS="sdkconfig.defaults.otbr" set-target esp32s3
idf.py menuconfig
"Enable matter commissioner" activated.
idf.py build

The example compiles whenever "Enable matter commissioner" is not activated, however it is important that the border router is able to commission and control devices without the need for ./chip-tool for my application.

shubhamdp, 2024-02-09:
Ohh, `CONFIG_ESP_MATTER_COMMISSIONER_ENABLE=n` is [disabled](https://github.com/espressif/esp-matter/blob/main/examples/controller/sdkconfig.defaults.otbr#L58) by default in `sdkconfig.defaults.otbr`.

Let me check and get back.

nichonaugle, 2024-02-15:
Any updates on the commissioner for esp-matter? @shubhamdp @jakabernat @dhrishi

wqx6, 2024-02-19:
@nichonaugle Sorry for the late reply. Please enable controller build `CONFIG_ENABLE_CHIP_CONTROLLER_BUILD=y` in menuconfig if you want to build a commissioner.

nichonaugle, 2024-02-25:
@wqx6 Thank you, that allowed it to compile properly. Would any other settings need to be changed in menuconfig to support commissioning over thread (i.e. matter over thread)?

wqx6, 2024-02-26:
Do you mean commissioning Matter-over-Thread end devices with the commissioner(enabling Thread BR)? If yes, I think current menuconfig setting is enough for that.

dhrishi, 2024-04-06:
@nichonaugle Please close this issue if resolved

closed 2024-04-07 by nichonaugle