espressif/esp-matter #836: Apple Home App, 8 Simple Light Endpoint, Slow response when "Turn Off All Lights"

ronny-antoon, 2024-02-12:
I have ESP32 board with 8 Relays, I had follow the simple light example and made 1 Node Endpoint, 1 Aggregator Endpoint , 8 Bridged node light endpoints.

All working as expected, but with a delay.

if i turn one light on, it took a 250 ms at least to response, and if i turn all lights on/off (8 lights). it take more than 1.5 seconds to response.

1) Why it take more than 250ms to response for one Command.?

2) When turning on all light, the relays start to turning on one by one, in a delay of 200 ms. Does this correct? Is their anything I can do to reduce this delays, or at least to get all commands in one shot.?

I am not an expert. Thanks to all replays.

shripad621git, 2024-02-14:
@ronny-antoon , can you please elaborate the need of using bridged node light endpoints and not simply eight lights on eight endpoints  in the above setup.

ronny-antoon, 2024-02-14:
> @ronny-antoon , can you please elaborate the need of using bridged node light endpoints and not simply eight lights on eight endpoints  in the above setup.

If i use eight lights on eight endpoints, it shows in apple home app one accessory(light) with eight services(on/off).

But i want eight accessories (light)

nichonaugle, 2024-02-15:
In the Apple home app, you are able to split up the 8 endpoints into individual "devices" from the GUI. Click on the settings of the device and you should see the option to do so. @ronny-antoon 
Not a problem with esp-matter as it is handled by Apple Home. Any latency could be due to WiFi. For improved speeds use Thread instead with chipset like H2 or C6.

ronny-antoon, 2024-02-15:
> In the Apple home app, you are able to split up the 8 endpoints into individual "devices" from the GUI. Click on the settings of the device and you should see the option to do so. @ronny-antoon
> Not [...]

1) Utilizing eight lights on eight endpoints prevents me from individually assigning them to rooms in the Apple Home app. Although I can technically split them, they end up stacked in the same room.

2) The latency issue I'm addressing is observed between the first "output" log and the last output in the log. It is not linked to the Wi-Fi connection.

jadhavrohit924, 2024-04-25:
@ronny-antoon Sorry for the delayed response. I have checked with the devkit led, by pressing boot button of devkit, LED on devkit takes 20ms to complete the action.
```
I (72669) app_driver: Toggle button pressed
I (72679) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00000000 is 0 **********
I (72689) led_driver_ws2812: led set r:0, g:0, b:0
I (75709) app_driver: Toggle button pressed
I (75719) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00000000 is 1 **********
I (75729) led_driver_ws2812: led set r:63, g:51, b:40
```

Since you have added relay and you are sending toggle command to node, it will take some ms to complete the action as application will process the command and take suitable action on driver. This is not a issue with esp-matter.

Please close the issue if you are satisfied with the answer.

ronny-antoon, 2024-04-25:
@jadhavrohit924 
Thank you,
Maybe you didn't understood me right.
My issue is not a button response.
My issue is homekit command response, when i click on "All Off" in apple home App, it takes more than 1.5 seconds to turn them off. And i see that every 200-250 ms one light turns off. And it turns one after one.

Esp32 with 8 relays, apple home app turns all off, it start "randomly" first relays turns off after 200ms, then the next relays turns off after another 200ms, then the third relay another 200ms etc.....
So it takes more than 1.5 seconds to turn off the 8 relays.!!!

jadhavrohit924, 2024-04-26:
> Esp32 with 8 relays, apple home app turns all off, it start "randomly" first relays turns off after 200ms, then the next relays turns off after another 200ms, then the third relay another 200ms etc.....

This is happening because every node is different and apple home send off command to each node sequentially (i.e. after completing action on one bulb it send off request to other node).

I have tried to make group of endpoints and send toggle command to a group using chip-tool. Groups also process each endpoint one by one.  ( Apple home doesn’t support groups )

ronny-antoon, 2024-04-27:
@jadhavrohit924 
So its an Apple Home "issue" and not a matter.
and till now their is no work around this problem.
Thank you a lot. i will close this issue.

closed 2024-04-27 by ronny-antoon