espressif/esp-matter #838: Issue with building light example in WSL2

HamidKetabdar, 2024-02-12:
I'm encountering a problem while trying to build the Matter Light example in WSL2. I've followed the instructions outlined on this page, https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html, to set up ESP IDF and ESP Matter but I get below error.

esp-idf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_gatt.h:885:13: error: expected unqualified-id before 'namespace'
  885 |     uint8_t namespace;
      |             ^~~~~~~~~

**Describe the bug**
A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id: fe4f9f6
- ESP-IDF Commit Id: 4b5b064
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6
- Device Logs (Please attach the log file):
[esp-matter-log.txt](https://github.com/espressif/esp-matter/files/14251821/esp-matter-log.txt)

- Host Machine OS: Windows 11 WSL2
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

sayondeep, 2024-02-13:
@HamidKetabdar Please change you IDF commit to tag v5.1.2 as mentioned [here](https://github.com/espressif/esp-matter/tree/main#supported-esp-idf-and-connectedhomeip-versions)

dhrishi, 2024-02-13:
Closing this issue. Please also see https://github.com/espressif/esp-matter/issues/830#issuecomment-1931897564

closed 2024-02-13 by dhrishi