espressif/esp-matter #841: simple light endpoint after reboot value not restored.

ronny-antoon, 2024-02-15:
**Describe the bug**
Simple light endpoint without any addition. just the root endpoint and the on_off_light endpoint.
No effect for the Config on_off.on_off, "this should make the start value".
After commission and all thing working as expected, Turn the light on. then reboot the device, the device start with Light off. but it should be On. as last value.

**Environment**
- ESP-Matter Commit Id: release/v1.2
- ESP-IDF Commit Id: v5.1.2
- SoC (eg: ESP32 or ESP32-C3): esp32
- Device Logs (Please attach the log file): 
[log esp mattter.txt](https://github.com/espressif/esp-matter/files/14292227/log.esp.mattter.txt)

- Host Machine OS: IOS 17.2.1
- Commissioner app and versions if present: IOS 17.2.1
- Commissioner's logs if present:

**Any additional details**
...

nichonaugle, 2024-02-15:
Is the starting value hardcoded into the example you have created or are you using the light example issued under the esp-matter repo?

jadhavrohit924, 2024-02-15:
@ronny-antoon Have you tried of_off_light with lighting feature enabled? StartUpOnOff attribute is optional and has conformance of LT i.e lighting feature.

ronny-antoon, 2024-02-15:
Simplest Code ever, @jadhavrohit924  @nichonaugle 

[log esp mattter.txt](https://github.com/espressif/esp-matter/files/14295082/log.esp.mattter.txt)

jadhavrohit924, 2024-02-15:
@ronny-antoon I have analysed your logs.  From the logs, you have toggled the light in-between thats why your light turned off and when you reboot it is stays off.

Can you try below steps and check if you are facing the issue:
1. Flash the device
2. Commission the device
3. Read value of OnOff attribute ( expected true )
4. Toggle the light
5. Read value of OnOff attribute ( expected false )
6. Reboot the device
7. Read value of OnOff attribute ( expected false )

ronny-antoon, 2024-02-15:
> @ronny-antoon I have analysed your logs.  From the logs, you have toggled the light in-between thats why your light turned off and when you reboot it is stays off.
>
>
>
> Can you try below steps and [...]

Thats not me,
Thats the problem,
If you are talking about the log starts with "chip[ZC....Toggle..."
I tried another time, Turned the light on. Reboot, then again it initiated with On, but turns off immediately with the initialization.

jadhavrohit924, 2024-02-19:
Could you please share the whole logs, which has everything right from flashing, turning on light, reboot, and state after reboot?

ronny-antoon, 2024-02-19:
@jadhavrohit924 
I have implemented a straightforward simple project featuring an on_off_light and an on_off_plugin_unit.
The program code can be found [here app_main.cpp.txt](https://github.com/espressif/esp-matter/files/14336629/app_main.cpp.txt) .

Having previously paired the device,
I initiated the project by executing the command 'idf.py flash monitor'. Once the device started up,
I accessed the Home App on Apple, toggled both the light and the plugin, both turned ON, and subsequently exited the terminal.
The corresponding output log for this sequence is available [here Flash monitor.txt](https://github.com/espressif/esp-matter/files/14336631/Flash.monitor.txt)

Subsequently, I ran the command 'idf.py monitor' without interacting with the device.
I patiently waited for the output log to stabilize, and the resulting log can be found [here monitor.txt](https://github.com/espressif/esp-matter/files/14336633/monitor.txt)

jadhavrohit924, 2024-02-20:
@ronny-antoon Thanks for the information, if you want to set OnOff to the previous state when device is supplied to power. You have to set StartUpOnOff attribute to null in your light config
```
on_off_light::config_t light_config;
light_config.on_off.lighting.start_up_on_off = nullptr;
on_off_light::create(node, &light_config, ENDPOINT_FLAG_NONE, nullptr);
```

ronny-antoon, 2024-02-20:
@jadhavrohit924 
Thanks for your efforts and time, i appresiate it.
I tried to make what you said about config, and still the same behavior.

`
    on_off_light::config_t light_config;
    light_config.on_off.lighting.start_up_on_off = nullptr;
    on_off_light::create(node, &light_config, ENDPOINT_FLAG_NONE, nullptr);

    on_off_plugin_unit::config_t plugin_unit_config;
    plugin_unit_config.on_off.lighting.start_up_on_off = nullptr;
    on_off_plugin_unit::create(node, &plugin_unit_config, ENDPOINT_FLAG_NONE, 
    nullptr);
`

ronny-antoon, 2024-02-21:
@jadhavrohit924 
Thank you a lot, I got the solution.
I don't know why,  but after Erase Flash, and re-Flash, It just solved the problem.
But NVM the goal achieved.  
Thank you.

closed 2024-02-21 by ronny-antoon