espressif/esp-matter #843: Missing `single` data type attribute value handling.

YOGYUI, 2024-02-16:
**Is your feature request related to a problem? Please describe.**
When updating attribute value which data type is `single` (single precision floating point), error occurred.

**Describe the solution you'd like**
I think following codes should be added in `esp_matter_attribute_utils.cpp`
```cpp
static esp_err_t get_attr_val_from_data(esp_matter_attr_val_t *val, EmberAfAttributeType attribute_type,
                                        uint16_t attribute_size, uint8_t *value,
                                        const EmberAfAttributeMetadata * attribute_metadata)
{
    switch (attribute_type) {
    /*
     * 
     */
    
    case ZCL_SINGLE_ATTRIBUTE_TYPE: {
        using Traits = chip::app::NumericAttributeTraits<float>;
        Traits::StorageType attribute_value;
        memcpy((float *)&attribute_value, value, sizeof(Traits::StorageType));
        if (attribute_metadata->IsNullable()) {
            if (Traits::IsNullValue(attribute_value)) {
                *val = esp_matter_nullable_float(nullable<float>());
            } else {
                *val = esp_matter_nullable_float(attribute_value);
            }
        } else {
            *val = esp_matter_float(attribute_value);
        }
        break;
    }
}
```

**Additional context**
I found that some attribute data types in esp-matter are not comply with matter specification.
Is there any reason?

[Example]
Cluster: Carbon Dioxide Concentration Measurement (id: 0x040C)
Attribute: Measured Value (id: 0x0000)
[image]
specification: `single` (nullable)
esp-matter code: `uint16_t`
```cpp
namespace carbon_dioxide_concentration_measurement {
namespace attribute {
attribute_t *create_measured_value(cluster_t *cluster, nullable<uint16_t> value)
{
    return esp_matter::attribute::create(cluster, CarbonMonoxideConcentrationMeasurement::Attributes::MeasuredValue::Id,
					 ATTRIBUTE_FLAG_NULLABLE, esp_matter_nullable_uint16(value));
}
```
Almost all single type attributes (like `MeasuredValue`, `MinMeasuredValue`, `MaxMeasuredValue`) of concentration measurement clusters are implemented as integer.

dhrishi, 2024-02-26:
@YOGYUI Thanks for reporting the issue and the probable solution. We will soon have the fix available on Github

YOGYUI, 2024-02-26:
@dhrishi Thanks for reply. 
Last week, I forked repo's main branch and created pull request to solve this issue. ([CON-1056](https://github.com/espressif/esp-matter/pull/848))
I'm now waiting for code review :)

closed 2024-04-02 by YOGYUI