espressif/esp-matter #845: run esp-matter-main\examples\light example failed

wangwenxue1989, 2024-02-19:
**Describe the bug**
I modify esp-matter-main\examples\light example，just add：
#define BSP_LED_NUM 1
Then compile with errors.

**Environment**
- ESP-Matter Commit Id: main
- ESP-IDF Commit Id: v5.1.2
- SoC (eg: ESP32 or ESP32-C3):ESP32-C3)
- Device Logs (Please attach the log file):
[error log.txt](https://github.com/espressif/esp-matter/files/14328465/error.log.txt)
- Host Machine OS:Ubuntu
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

would somebody can help? Thanks!

jadhavrohit924, 2024-02-19:
@wangwenxue1989 This should not happen, error says SET_HUE and SET_SATURATION are not defined. Could you please check if you missed to include some header in your project? Or share the patch that you have created on example/light.

dhrishi, 2024-03-05:
@wangwenxue1989 Please share the details requested. Or please close the issue if it is working fine

dhrishi, 2024-03-26:
Please retry with the latest codebase. In case of any issues, please re-open by providing the requested details.

closed 2024-03-26 by dhrishi