espressif/esp-matter #846: Using Matter light example to control onboard addressable LED for esp32c6

julesrodrigues11, 2024-02-20:
I was working with the light example available on the esp-matter repository on the esp32c6 board. I have been trying to make it such that the power button would toggle the addressable LED on board the esp32c6. To this extent, I studied the blink example that is available in the esp-idf repository, and was working on merging the code together. However, I wasn't quite achieving the result I wanted. How would I go about doing this? Many thanks for your time in advance.

espzav, 2024-02-21:
Yes, this is right settings:
```
# ESP32-C6-DevKitM-1 Settings
# Buttons
CONFIG_BSP_BUTTONS_NUM=1
CONFIG_BSP_BUTTON_1_TYPE_GPIO=y
CONFIG_BSP_BUTTON_1_GPIO=9
CONFIG_BSP_BUTTON_1_LEVEL=0
# LEDs
CONFIG_BSP_LEDS_NUM=1
CONFIG_BSP_LED_TYPE_RGB=y
CONFIG_BSP_LED_RGB_GPIO=8
CONFIG_BSP_LED_RGB_BACKEND_RMT=y
```

dhrishi, 2024-03-12:
The fix is merged to Github with https://github.com/espressif/esp-matter/commit/bed4aa92e339f28a6419a9d438d3837b30bd51fa. Closing this issue. @espzav Can you please check the above question around the documentation. Thanks!

closed 2024-03-12 by dhrishi

espzav, 2024-03-26:
@julesrodrigues11 Here is documentation for Generic BSP, which is used there: https://github.com/espressif/esp-bsp/tree/master/bsp/esp_bsp_generic