espressif/esp-matter #847: Problems with matter when running parallel to webpage hosting

S-A-M-J, 2024-02-22:
**Describe the bug**
esp fails to initialize and use advertiser 

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC: ESP32C3
- Device Logs (Please attach the log file):
- Host Machine OS: MacOs

Hi, i have code that hosts a webpage on a network in STA mode (esp is client in network and is hosting a webpage). On this webpage there is a button that if clicked turns matter on and makes the device discoverable and and controllable through matter while keeping the input possibility from the hosted webpage. Now at this point there are two tasks running. One with the webpage stuff with idlestatus priority and another one that starts the matter part that has priority idle + 1.
When i start matter, i get the below added logs. With the chip tool i tried to connect to the esp with onnetwork pairing but it seems the device is not even found (nodeid and passcode are guaranteed correct). When using amazon alexa to discover the device, the device is found but Alexa tells me to enter wifi creds in a loop while never fully connecting. (2nd logs).

The code to start matter i used is `err = esp_matter::start(app_event_cb);`

Maybe someone here has an idea what exactly is going wrong or what i can change. The matter code is 99% based on the light example from esp-matter.

[1.pdf](https://github.com/espressif/esp-matter/files/14380361/1.pdf)
[2.pdf](https://github.com/espressif/esp-matter/files/14380360/2.pdf)

wqx6, 2024-02-23:
Hi, the logs show a LwIP error ERR_USE(-8). It means that the mDNS port(5353) of minimal mDNS was used by another pcb. Could you check whether you are using the mdns component https://github.com/espressif/esp-protocols/tree/master/components/mdns? If yes and you are not going to remove it, please disable minimal mDNS and use platform mDNS instead.
```
CONFIG_USE_MINIMAL_MDNS=n
```

S-A-M-J, 2024-02-25:
Perfect thank you. Yes i was using the mdns component for the webpage hosting. Turning off the minimal dns actually got rid of the issue. I'm assuming by platform mdns you are referring to the mentioned component? And how did you connect the error code to mdns? i wasnt able to find a list of the error codes. 
However now the esp seems to be impossible to find as a matter device, though there are no more errors. Could this be related to turning off minimal mdns or is it as i suspect unrelated?

wqx6, 2024-02-26:
> I'm assuming by platform mdns you are referring to the mentioned component

Yes, the platform mdns is using the mdns component I mentioned.

> And how did you connect the error code to mdns? i wasnt able to find a list of the error codes.

The logs say `E (29017) chip[DIS]: Failed to initialize advertiser: 3000008`. It is from the minimal mDNS part and the [error code](https://github.com/project-chip/connectedhomeip/blob/master/src/lib/core/CHIPError.h#L306-L315) is `0x3000008`. The range is 0x3 so it's a [LwIP error](https://github.com/project-chip/connectedhomeip/blob/master/src/lib/core/CHIPError.h#L90). And the LwIP error code is -8(ERR_USE). Normally it happens when a port is duplicately used. But 5353 is a special port for mDNS. So I guess that you are using mdns component.

> However now the esp seems to be impossible to find as a matter device, though there are no more errors. Could this be related to turning off minimal mdns or is it as i suspect unrelated?

Maybe you can try to resolve the operational node service and see whether the service is advertised by your device? Looks like there are some issues on the order of mdns component APIs calling.

S-A-M-J, 2024-02-28:
Turns out the device was actually visible. Sry my mistake. Thank you for the explanation.
 Everything works now but after 2-3 minutes the webserver will suddenly stop answering while the matter interface works just fine. I dont get this issue when i dont start matter and the webserver works indefinitely. Does anybody what the problem could be here? There are no errors given.

S-A-M-J, 2024-02-28:
So some testing showed that the webserver only stops responding if i don't message it for more than a minute. If i contiously send messages with less time than 1 min between them it will keep responding.

S-A-M-J, 2024-02-29:
Unfortunately not closed yet. While keeping the connection alive with pings works, once i close the web tab and open a new to connect to the webserver it is unresponsive again. It must have something to do with sessions being closed and unable to be reopened again but unfortunately i cant seem to find out exactly what.

S-A-M-J, 2024-03-03:
So i found out that there is no way to keep the hostname.

closed 2024-03-03 by S-A-M-J