espressif/esp-matter #851: Multi Ecosystem Commissioning: Status Corruption

nichonaugle, 2024-02-25:
After commissioning successful to a Google Nest Hub 2 via Thread, the light example functioned properly before suddenly disconnecting days later. It was unable to reconnect after multiple power cycles and issue was replicated onto 3 more light example devices commissioned via Thread with the same results. 

After Google proved to be unreliable, I attempted to transfer devices to Apple HomePod via Thread. 

2/4 devices successfully commissioned and remain functional, with the remaining two posing the prior below. 

I am unaware if the issue stems from Google Home or if the commissioning structure is not cleanly erased when a factory reset is instantiated. 

**Environment**
- ESP-Matter Commit Id: most recent
- ESP-IDF Commit Id: v5.1.2
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6
- Device Logs (Please attach the log file):
- Host Machine OS: MacOS
- Commissioner app and versions if present: Apple Home on IPhone 15
- Commissioner's logs if present:
[image]

I factory reset the devices multiple times with the google nest hub plugged in and unplugged with the same result as shown in the image above. I am concerned it is an issue with improper device factory reset, where credentials are not fully erased by the esp32c6. 
...

VaishaliAvhale, 2024-02-26:
Please attach device logs file

VaishaliAvhale, 2024-03-12:
I observed uncertain behavior with Google Home while putting it on for a long duration. I will check with Google and provide an update here soon.

VaishaliAvhale, 2024-07-02:
Last month I set the log duration for a complete week and I haven't faced this issue. Please check at your end and close if it is resolved.

dhrishi, 2024-07-26:
Closing this now. If you are still facing the issue, please provide the requested logs and re-open

closed 2024-07-26 by dhrishi