espressif/esp-matter #852: advertising additional mDNS services

albrownwood, 2024-02-28:
Hello,
what is the recommended way for advertising additional mdns service without perturbing Matter dns advertising behavior?
I have tried deactivating the "minimal mDNS that ships with CHIP" and use the `mdns_service_add(NULL, "_aaaa", "_tcp", 80, NULL, 0);` API but that returns an error ESP_ERR_INVALID_ARG.

wqx6, 2024-02-28:
Is the mdns server host name set before you call `mdns_sevice_add()`? You could call `mdns_hostname_get()` to check whether the hostname is set.

albrownwood, 2024-02-28:
@wqx6 I tried calling get_hostname and that returned an error ESP_ERR_INVALID_STATE. This occured when I place the mdns calls inside the kServerReady event callback.

I played with moving that mdns init code in other callbacks, for example when placed in kDnssdInitialized, I get same errors.

When I place the mdns code in kInternetConnectivityChange it sometimes work and I can retrieve the current hostname and add a service but sometimes it doesn't. So it looks like this is not a reliable event that can be used to know the mdns layer is ready to interact with. I would have expected kDnssdInitialized to be such reliable event.

So the I could probably make it work by waiting a long time after boot to be sure mdns layer is initialized but that's not pretty... ideally there should be an event callback that can reliably indicates the dns layer is ready to be interacted with.

wqx6, 2024-02-29:
`get_hostname()` returning ESP_ERR_INVALID_STATE means that the mdns sever is not initialized or the hostname is never set.  In platform mDNS implement of Matter, we will only set the host when [publishing service](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/WiFiDnssdImpl.cpp#L164). But the kDnssdInitialized will be posted after [DnssdInit](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/WiFiDnssdImpl.cpp#L133) and the hostname is not set at that time. Since the matter will always use MAC address as the hostname and our platform mDNS will remap all old hostname's services to new hostname, you can call `mdns_hostname_set()` with the MAC address before you call the `add_service` APIs.
```
if (mdns_hostname_get() != ESP_OK) {
    mdns_hostname_set(MAC_ADDRESS_STRING);
    mdns_service_add();
}
```
Note that you should add these codes after receiving `kDnssdInitialized` event (`mdns_init()`).

albrownwood, 2024-03-01:
thanks for the resolution!

closed 2024-03-01 by albrownwood