espressif/esp-matter #853: assert failed: igmp_joingroup_netif (Required to lock TCPIP core functionality!)

PhLuReh, 2024-02-28:
**Describe the bug**
After updating esp-matter to f511d226420f56a0565f7b147ebc399f88afc3ce with all its referenced submodules, the device is facing an assertion error in the LWIP while joining MDNS groups. This happens to a to-be-commissioned device after Wifi connection is established.

Could you assist in finding the correct build framework or even a fix?

```
 (202073) chip[DL]: Updating advertising data
I (202203) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (202963) esp_netif_handlers: sta ip: 192.168.1.127, mask: 255.255.255.0, gw: 192.168.1.1
I (202963) chip[DL]: IP_EVENT_STA_GOT_IP
I (202963) chip[DL]: IPv4 address changed on WiFi station interface: 192.168.1.127/255.255.255.0 gateway 192.168.1.1
I (202973) chip[DL]: IPv4 Internet connectivity ESTABLISHED
I (202983) app_main: Interface IP Address changed
I (202983) chip[DIS]: Updating services using commissioning mode 0

assert failed: igmp_joingroup_netif /IDF/components/lwip/lwip/src/core/ipv4/igmp.c:495 (Required to lock TCPIP core functionality!)

Backtrace: 0x40081afe:0x3ffe8eb0 0x400927f9:0x3ffe8ed0 0x400994a1:0x3ffe8ef0 0x401575d9:0x3ffe9010 0x40129484:0x3ffe9030 0x40129015:0x3ffe9060 0x40118ff4:0x3ffe9090 0x40107369:0x3ffe9100 0x40105aae:0x3ffe9130 0x400e4096:0x3ffe9150 0x400e412a:0x3ffe9170 0x400e1109:0x3ffe9190 0x401c62a5:0x3ffe91b0 0x401147ad:0x3ffe91d0 0x40114acc:0x3ffe91f0 0x40114b01:0x3ffe9260 0x40094fad:0x3ffe9280

  #0  0x40081afe:0x3ffe8eb0 in panic_abort at /opt/espressif/esp-idf/components/esp_system/panic.c:452
  #1  0x400927f9:0x3ffe8ed0 in esp_system_abort at /opt/espressif/esp-idf/components/esp_system/port/esp_system_chip.c:84
  #2  0x400994a1:0x3ffe8ef0 in __assert_func at /opt/espressif/esp-idf/components/newlib/assert.c:81
  #3  0x401575d9:0x3ffe9010 in igmp_joingroup_netif at /opt/espressif/esp-idf/components/lwip/lwip/src/core/ipv4/igmp.c:495 (discriminator 1)
  #4  0x40129484:0x3ffe9030 in chip::Inet::UDPEndPointImplLwIP::IPv4JoinLeaveMulticastGroupImpl(chip::Inet::InterfaceId, chip::Inet::IPAddress const&, bool) at build/esp-idf/chip/../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/inet/UDPEndPointImplLwIP.cpp:482 (discriminator 1)
  #5  0x40129015:0x3ffe9060 in chip::Inet::UDPEndPoint::JoinMulticastGroup(chip::Inet::InterfaceId, chip::Inet::IPAddress const&) at build/esp-idf/chip/../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/inet/UDPEndPoint.cpp:146
  #6  0x40118ff4:0x3ffe9090 in mdns::Minimal::ServerBase::Listen(chip::Inet::EndPointManager<chip::Inet::UDPEndPoint>*, mdns::Minimal::ListenIterator*, unsigned short) at build/esp-idf/chip/../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/dnssd/minimal_mdns/Server.cpp:218
  #7  0x40107369:0x3ffe9100 in chip::Dnssd::GlobalMinimalMdnsServer::StartServer(chip::Inet::EndPointManager<chip::Inet::UDPEndPoint>*, unsigned short) at build/esp-idf/chip/../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/dnssd/MinimalMdnsServer.cpp:69
  #8  0x40105aae:0x3ffe9130 in chip::Dnssd::(anonymous namespace)::AdvertiserMinMdns::Init(chip::Inet::EndPointManager<chip::Inet::UDPEndPoint>*) at build/esp-idf/chip/../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/dnssd/Advertiser_ImplMinimalMdns.cpp:350
  #9  0x400e4096:0x3ffe9150 in chip::app::DnssdServer::StartServer(chip::Dnssd::CommissioningMode) at /opt/espressif/esp-matter/connectedhomeip/connectedhomeip/src/app/server/Dnssd.cpp:376
  #10 0x400e412a:0x3ffe9170 in chip::app::DnssdServer::StartServer() at /opt/espressif/esp-matter/connectedhomeip/connectedhomeip/src/app/server/Dnssd.cpp:343
  #11 0x400e1109:0x3ffe9190 in esp_matter::device_callback_internal(chip::DeviceLayer::ChipDeviceEvent const*, int) at /opt/espressif/esp-matter/components/esp_matter/esp_matter_core.cpp:901
  #12 0x401c62a5:0x3ffe91b0 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToApplication(chip::DeviceLayer::ChipDeviceEvent const*) at build/esp-idf/chip/../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.ipp:336
  #13 0x401147ad:0x3ffe91d0 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at build/esp-idf/chip/../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.ipp:301
  #14 0x40114acc:0x3ffe91f0 in chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at build/esp-idf/chip/../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:505
      (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at build/esp-idf/chip/../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:257
  #15 0x40114b01:0x3ffe9260 in chip::DeviceLayer::PlatformManager::RunEventLoop() at build/esp-idf/chip/../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:405
      (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at build/esp-idf/chip/../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:280
  #16 0x40094fad:0x3ffe9280 in vPortTaskWrapper at /opt/espressif/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```
here is a longer log:
[20240228_DUT-log.txt](https://github.com/espressif/esp-matter/files/14435316/20240228_DUT-log.txt)

**Environment**
- ESP-Matter Commit Id: f511d226420f56a0565f7b147ebc399f88afc3ce
- ESP-IDF Commit Id: e088c3766ba440e72268b458a68f27b6e7d63986
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file): 
[20240228_DUT-log-long.txt](https://github.com/espressif/esp-matter/files/14435321/20240228_DUT-log-long.txt)
- Host Machine OS: `Linux version 6.1.29-1-MANJARO (builduser@fv-az292-908) (gcc (GCC) 12.2.1 20230201, GNU ld (GNU Binutils) 2.40) #1 SMP PREEMPT_DYNAMIC Wed May 17 14:00:55 UTC 2023`
- Commissioner app and versions if present: n/a
- Commissioner's logs if present: n/a

**Any additional details**

As I suspect there is some dependency issue, here is a full list of the connectedhomeip subodule hashes:
```
cd /opt/espressif/esp-matter/; git submodule foreach --recursive "git log | head -n 1"
Entering 'connectedhomeip/connectedhomeip/examples/common/QRCode/repo'
commit 8518684c0f33d004fa93971be2c6a8eca3167d1e
Entering 'connectedhomeip/connectedhomeip/examples/common/m5stack-tft/repo'
commit 3344a8105f8dfecb95832f369733cf6f863a1662
Entering 'connectedhomeip/connectedhomeip/third_party/boringssl/repo/src'
commit c8d31372f7342707d8c6e40c814ce1b64fe36086
Entering 'connectedhomeip/connectedhomeip/third_party/bouffalolab/repo'
commit 7ca85fd77663ab6e13ce8481593474ea38ad4bb2
Entering 'connectedhomeip/connectedhomeip/third_party/bouffalolab/repo/components'
commit 28387939c3c7ce1273f14d00bca4ba73cfe95d9f
Entering 'connectedhomeip/connectedhomeip/third_party/bouffalolab/repo/customer_app'
commit 5454c795a68d06ced29c13cfafccf892dee8bb33
Entering 'connectedhomeip/connectedhomeip/third_party/bouffalolab/repo/docs'
[... 223 lines cut ...]
Entering 'connectedhomeip/connectedhomeip/third_party/ot-br-posix/repo/third_party/cJSON/repo'
commit cf97c6f066d81fdbba4ef722cfd327bbbba2365c
Entering 'connectedhomeip/connectedhomeip/third_party/ot-br-posix/repo/third_party/http-parser/repo'
commit 4f15b7d510dc7c6361a26a7c6d2f7c3a17f8d878
[... cut ...]
Entering 'connectedhomeip/connectedhomeip/third_party/silabs/wiseconnect-wifi-bt-sdk'
commit 2be7a5b87fd3be0111cd8bc09585219e104f26f3
Entering 'connectedhomeip/connectedhomeip/third_party/simw-top-mini/repo'
commit be0dfb64b28dea3509425755e3b4b5b35c38d327
Entering 'connectedhomeip/connectedhomeip/third_party/ti_simplelink_sdk/repo_cc13xx_cc26xx'
commit cac6a455adffdcbc2e6c9499a0068c3d8c44b994
Entering 'connectedhomeip/connectedhomeip/third_party/ti_simplelink_sdk/repo_cc32xx'
commit 13532baa2947440ced2f2abefd6a57b478f5100d
```

PhLuReh, 2024-02-28:
Btw, I found this issue https://github.com/espressif/esp-idf/issues/2580 for the idf, but it was from 2018 and I think this one should be fixed.

dhrishi, 2024-02-28:
@PhLuReh Can you please upload your sdkconfig file? Is this with the standard light example (without any changes)?
Also, does this occur everytime or in a random manner?

PhLuReh, 2024-02-29:
this is my [sdkconfig](https://github.com/espressif/esp-matter/files/14446660/sdkconfig.txt)
The problem is persistent. Everytime I try to commission the device with the chip-tool this error occurs.

I tried to run the light app from the examples, but I was not able to commission to the chip-tool. (I was using a ESP32S3 in favour of my ESP32 from my product im developing). The device is only getting connected via BLE and responds to chip-tool, but no Wifi-Credentials are set nor connected to the AP.

dhrishi, 2024-02-29:
@PhLuReh Looks like some of the sdkconfig options do not match the Matter v1.1 codebase. Were you at some point working on v1.2 or main branch of Matter with the same example? If yes, can you delete your sdkconfig file and give it a try

```
$ cd /path/to/your_example
$ rm -rf sdkconfig sdkconfig.old build/
$ idf.py menuconfig #If required to change any options
$ idf.py erase-flash build flash monitor
```

closed 2024-02-29 by PhLuReh

dhrishi, 2024-02-29:
Your attached sdkconfig has 
```
CONFIG_LWIP_TCPIP_CORE_LOCKING=y
CONFIG_LWIP_CHECK_THREAD_SAFETY=y
```
These options are enabled in Matter v1.2 and not in Matter v1.1. Now, when you used these options with v1.1, the corresponding code related to it is not present on the v1.1 branch and hence the errors.

0952594107, 2024-11-26:
1. 0952594107

0952594107, 2024-11-26:
pgslot100%