espressif/esp-matter #854: Enquiries on BLE stack used in the esp-matter commissioning steps

eme-bennytang, 2024-02-29:
**Describe the question/query that you have**
To my knowledge, the BLE is only used in the commisioning steps in advertisements, CASE connection.
I wanted to confirm what is the exact BLE stack version (4.2?  5.1?) that is used?

From my SBOM created from the project, I can see Nimble component is using BLE stack 5.1
[image]

While I got comments from customer when they are doing the penetration-test, saying its likely to be using BLE 4.2.
Furthermore, I asked the esspressif local support and they said in the commssioning steps, BLE 4.2 is used to ensure compability from older devices. 

Does this mean in the CSA attestation we need to declare the protocol version for BLE as v4.2 instead of v5.1?
[image]

Can anyone share more knowledge/details on the BLE stack used, as well as the protocol version we should declare in the CSA attestation form?

**Additional context**
esp-idf: v5.1.2
esp-matter: #2f692ac1
MCU: esp32s3

rahult-github, 2024-03-01:
Hi @eme-bennytang 

So first, as mentioned in bluetooth specification , bluetooth specifications need to be backward compatible. This implies if BLE version is 5.1 , it also supports the previous ( 5.0 / 4.2 ) versions.

Commissioning in matter uses legacy adv and this ensures that it works with 4.2 devices . 

Hence, mentioning that stack / protocol is 5.1 doesn't imply that we do not work with devices with 4.2 stack. So, mentioning 5.1 BLE stack / protocol version should be ok , since the functionality used in the procedures ensure we do not cause any IOT issues.

dhrishi, 2024-03-12:
@eme-bennytang Please close the issue if it answers your question

dhrishi, 2024-03-18:
Closing. Feel free to reopen in case of any further questions

closed 2024-03-18 by dhrishi