espressif/esp-matter #860: When is the support for motion sensors

itatboy, 2024-03-04:
**Is your feature request related to a problem? Please describe.**
Description of what the problem is.

**Describe the solution you'd like**
Description of what you want to happen.
Support motion sensors

**Describe alternatives you've considered**
Description of any alternative solutions or features you've considered.

**Additional context**
I couldn't find any motion sensors in the supported device list
[image]

jadhavrohit924, 2024-03-04:
@itatboy There is no such device type as of now as per matter spec.

itatboy, 2024-03-04:
Thank you for your reply,

As far as I know, Apple Home comes with a Motion sensor accessory. Is there a similar device type for the subject. I want to implement the corresponding Motion Sensor accessory through matter, so that the matter device will be a Motion Sensor when added to the Home App？

I also noticed that:

There are already Matter based devices on the market that are recognized as Motion Sensors when added to the Home App, and it is currently unclear whether they are using ESP solutions.

jadhavrohit924, 2024-03-04:
Thanks for the information, there is no any such dedicated device types. People are using above (attached list by you) sensor device types to make motion sensor.

itatboy, 2024-03-04:
https://www.apple.com/shop/product/HQWK2ZM/A/Eve%20Motion%20(Matter)%20Sensor

The above is the official link from Apple, which displays a Motion sensor based on Matter implementation. May I ask which device type of Matter corresponds to a Motion sensor?

Thank you very much for any information provided

jadhavrohit924, 2024-03-04:
I don’t know what they have implementet but it looks like a `occupancy sensor`. Here is the list of [supported device type](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#testing-your-matter-device-with-apple-home) by apple and it doesn’t have `motion sensor`

jonsmirl, 2024-03-07:
Motion sensor is an occupancy sensor. 
[image]

itatboy, 2024-04-01:
@jonsmirl Hi,

Thank you very much for providing the information, it is very useful. Now I know how to implement sports accessories.
thanks

closed 2024-04-01 by itatboy

itatboy, 2024-04-04:
@jonsmirl 

I'm sorry, after verification, I can only reopen this issue.Here are some of my progress:
I use occupancy sensors to implement motion sensors. The code is as follows:
[image]
[image]

When accessories are added to Homekit, they are recognized as human sensors,It's not a motion sensor
[image]

https://csa-iot.org/csa_product/eve-motion-3/
When the EVE motion developed based on Matter was added to the Home App, it was identified as a motion sensor. I reviewed its compliance certification documents and found that it supports the PIROccupiedToUnocupiedDelay attribute, while the esp matter does not. I am not sure if this is causing the device recognition error. Can you provide a solution to the problem? Thank you.
[image]
[image]

This is the effect I want, the accessory has been identified as a motion sensor:
[image]

jonsmirl, 2024-04-04:
I think Homekit is just being more generic in categorizing them as Presence sensors.  All motion sensors are presence sensors, but not all presence sensors are motion sensors. For example there are heat source sensors which can tell if there are warm bodies in the room even if they aren't moving.  There are also radar based sensors which can see a heart beat even if you aren't moving.

itatboy, 2024-04-04:
You are right, all motion sensors are presence sensors. It is not clear how to make the Home App recognize the accessory as a motion sensor. It is not clear its judgment mechanism. I still need to try it. Thank you. @jonsmirl

itatboy, 2024-04-05:
@jonsmirl Hi,

I noticed an interesting phenomenon, is it a bug?
In the code, set occupancy_sensor_type=0x01, occupancy_sensor_type_bitmap=0x02. After burning to the program, start the log information occupancy_sensor_type=0x00, occupancy_sensor_type_bitmap=0x01

[image]

jonsmirl, 2024-04-05:
Check out this file:
https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/occupancy-sensor-server/occupancy-sensor-server.cpp

This might be getting called when it is not supposed to:
https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/occupancy-sensor-server/occupancy-sensor-server.cpp#L85
You can try putting a breakpoint on it and see if it is getting called.

It appears to me something is overwriting your initialization when it shouldn't.

itatboy, 2024-04-05:
@jonsmirl 
You're right, it did run void emberAfOccupancySensingClusterServerInitCallback(EndpointId endpoint) ---> void emberAfPluginOccupancyClusterServerPostInitCallback(EndpointId endpoint).
The callback function is registered in namespace occupancy sensing cluster_t * create (endpoints, config_t * config, uint8_t flags).

What situation would cause the callback function to run? Thank you