espressif/esp-matter #861: assert failed: xEventGroupSetBits event_groups.c:559 (xEventGroup) after BLE is deinitialized

law-ko, 2024-03-04:
**Describe the bug**

```
I (4498) chip[DL]: CHIPoBLE advertising stopped
I (4868) fleet_prov_nvs_helper: Reading KEY from aws_certkey NVS namespace successful!
I (4868) chip[IM]: No subscriptions to resume
I (4868) app_main: BLE is deinitialized

assert failed: xEventGroupSetBits event_groups.c:559 (xEventGroup)

Backtrace: 0x40081c96:0x3ffeed30 0x400943c9:0x3ffeed50 0x40098f0d:0x3ffeed70 0x4020f312:0x3ffeee90 0x400d92f2:0x3ffeeeb0 0x4020776d:0x3ffeeed0 0x4015c08d:0x3ffeeef0 0x4015c3b0:0x3ffeef10 0x4015c3e5:0x3ffeef80 0x400948ca:0x3ffeefa0
0x40081c96: panic_abort at /home/espuser/esp/esp-idf/components/esp_system/panic.c:452
0x400943c9: esp_system_abort at /home/espuser/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x40098f0d: __assert_func at /home/espuser/esp/esp-idf/components/newlib/assert.c:81
0x4020f312: xEventGroupSetBits at /home/espuser/esp/esp-idf/components/freertos/FreeRTOS-Kernel/event_groups.c:560 (discriminator 1)
0x400d92f2: app_event_cb(chip::DeviceLayer::ChipDeviceEvent const*, int) at /home/espuser/esp-git/esp-matter/examples/im/outlet/build/../main/app_main.cpp:117 (discriminator 13)
0x4020776d: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToApplication(chip::DeviceLayer::ChipDeviceEvent const*) at /home/espuser/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.ipp:336
0x4015c08d: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/espuser/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.ipp:301
0x4015c3b0: chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/espuser/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:503
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at /home/espuser/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:257
0x4015c3e5: chip::DeviceLayer::PlatformManager::RunEventLoop() at /home/espuser/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:403
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /home/espuser/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:280
0x400948ca: vPortTaskWrapper at /home/espuser/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

**Environment**
- ESP-Matter Commit Id : release/v1.2 0b515943fffd35243bc2d3d124e6dafdfae3cb11
- ESP-IDF Commit Id: ESP-IDF v5.1.2-dirty
- SoC (eg: ESP32 or ESP32-C3): ESP32-PICO-MINI-02
- Device Logs (Please attach the log file):
- Host Machine OS: Linux
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
This happens randomly and not sure which part failed.

dhrishi, 2024-03-08:
@law-ko Even though it happens randomly, can you please provide a set of steps that you followed that caused the issue?

law-ko, 2024-03-11:
@dhrishi This happens at the boot of the device. When the device is commissioned and performed a reboot.

shubhamdp, 2024-04-03:
@law-ko From the backtrace, I can only figure out that it is in the event handler and its because you are trying to set the group bits that are being used by the FreeRTOS kernel. Please take a look at https://github.com/espressif/esp-idf/blob/v5.1.2/components/freertos/FreeRTOS-Kernel/event_groups.c#L557-L560

esp-matter SDK do not use `xEventGroupSetBits` anywhere, If your application is using them, please take a look at this [API guide](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/api-reference/system/freertos.html#event-group-api) for more info.

dhrishi, 2024-04-11:
@law-ko Can you please check the above comment?

shawty, 2024-12-02:
Not the original poster here, but I just been hit by the same thing:

This is my entire output log:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (27) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (27) boot: compile time Dec  2 2024 21:34:26
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=12d28h ( 77096) map
I (113) esp_image: segment 1: paddr=00022d50 vaddr=3fc94600 size=02b04h ( 11012) load
I (116) esp_image: segment 2: paddr=0002585c vaddr=40374000 size=0a7bch ( 42940) load
I (128) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=3efd8h (258008) map
I (175) esp_image: segment 4: paddr=0006f000 vaddr=4037e7bc size=05e00h ( 24064) load
I (187) boot: Loaded app from partition at offset 0x10000
I (187) boot: Disabling RNG early entropy source...
I (199) cpu_start: Multicore app
I (208) cpu_start: Pro cpu start user code
I (208) cpu_start: cpu freq: 160000000 Hz
I (209) app_init: Application information:
I (211) app_init: Project name:     cb-device-firmware
I (217) app_init: App version:      1
I (222) app_init: Compile time:     Dec  2 2024 21:51:56
I (228) app_init: ELF file SHA256:  ae3099f31...
I (233) app_init: ESP-IDF:          v5.3.1
I (238) efuse_init: Min chip rev:     v0.0
I (242) efuse_init: Max chip rev:     v0.99
I (247) efuse_init: Chip rev:         v0.2
I (252) heap_init: Initializing. RAM available for dynamic allocation:
I (259) heap_init: At 3FC97E20 len 000518F0 (326 KiB): RAM
I (265) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (272) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (278) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (285) spi_flash: detected chip: boya
I (288) spi_flash: flash io: dio
I (293) sleep: Configure to isolate all GPIO pins in sleep state
I (299) sleep: Enable automatic switching of GPIO sleep configuration
I (307) main_task: Started on CPU0
I (337) main_task: Calling app_main()
I (337) Main: APP_MAIN executing.
I (337) Main: Free memory: 378684 bytes
I (337) Main: IDF version: v5.3.1
I (347) Main: Starting ModemHandler Task...
System Date is: 01/01/1970
System Time is: 00:00:00
Get time result : 0
myTime->tv_sec : 0
myTime->tv_usec : 144777
I (357) APP_MAIN: Main loop 1 second tick
System Date is: 01/01/1970
[... 92 lines cut ...]

```
/*
 * TaskEventBits.h
 *
 *  Created on : 2 Dec 2024
 *  Author     : P.J.Shaw
 *  Purpose    : Hold's system wide DEFINES for the ESP-IDF event group system, allowing event based
 *             : Task syncronisation, all TASK modules should include this header.
 */

#ifndef TASKEVENTBITS_H
#define TASKEVENTBITS_H

#define COMMS_AVAILABLE_BIT (1 << 0)
#define DEVICE_TIME_SET_BIT (1 << 1)
#define MQTT_AVAILABLE_BIT (1 << 2)

#include "freertos/FreeRTOS.h"

// Main application wide event group, uses flag definitions in 'TaskEventBits.h'
static EventGroupHandle_t ApplicationEvents;

#endif /* TASKEVENTBITS_H */
```

shawty, 2024-12-02:
I just read on another page, that "task groups" need to be created, so in lieu of that, I added :

```
extern "C" void app_main(void)
{
  ApplicationEvents = xEventGroupCreate(); // <--------------- THIS LINE
  
  ESP_LOGI("Main", "APP_MAIN executing.");
  ESP_LOGI("Main", "Free memory: %d bytes", (int)esp_get_free_heap_size());
  ESP_LOGI("Main", "IDF version: %s", esp_get_idf_version());
  
  ESP_LOGI("Main", "Starting ModemHandler Task...");
  AppModem = new ModemHandler();
  AppModem->StartTask();
```

To my app main file, but I still get the same core dump at boot time.

shawty, 2024-12-02:
one more update (After reading further):

I got it working.  Usually the case, for some reason, you can't fix it, then as soon as you a question asking for help, you sort it out.

ANYWAY, turns out that the event system is quite picky, I had to move the definition of my event group out of the header file, into my main task, then declare the header file as an "extern".  so in main.cpp:

```
// System TASK Module definitions, all defined in the 'tasks' folder
ModemHandler *AppModem; 

// Main application wide event group, uses flag definitions in 'TaskEventBits.h'
EventGroupHandle_t ApplicationEvents;

// Main APPLICATION Task entry point, this is the DEFAULT FreeRTOS task created
// by ESP-IDF to kick start the firmware.  All other tasks and modules shall be
// spawned from here.
extern "C" void app_main(void)
{
  ApplicationEvents = xEventGroupCreate();
  
  ESP_LOGI("Main", "APP_MAIN executing.");
  ESP_LOGI("Main", "Free memory: %d bytes", (int)esp_get_free_heap_size());
  ESP_LOGI("Main", "IDF version: %s", esp_get_idf_version());
  
  ESP_LOGI("Main", "Starting ModemHandler Task...");
  AppModem = new ModemHandler();
  AppModem->StartTask();
```

In the header file:

```
/*
 * TaskEventBits.h
 *
 *  Created on : 2 Dec 2024
 *  Author     : P.J.Shaw
 *  Purpose    : Hold's system wide DEFINES for the ESP-IDF event group system, allowing event based
 *             : Task syncronisation, all TASK modules should include this header.
 */

#ifndef TASKEVENTBITS_H
#define TASKEVENTBITS_H

#define COMMS_AVAILABLE_BIT (1 << 0)
#define DEVICE_TIME_SET_BIT (1 << 1)
#define MQTT_AVAILABLE_BIT (1 << 2)

#include "freertos/FreeRTOS.h"

// Main application wide event group, uses flag definitions in 'TaskEventBits.h' definition in main task.
extern EventGroupHandle_t ApplicationEvents;

#endif /* TASKEVENTBITS_H */
```

and then in my modem handler:

```
#include <the headerfile here>
.
.
.
.
  if (event_id == IP_EVENT_PPP_GOT_IP)
  {
    ESP_LOGI("ModemHandler", "PPP Connection has been created.");
    auto *event = (ip_event_got_ip_t *)event_data;
    ESP_LOGI("ModemHandler", "IP          : " IPSTR, IP2STR(&event->ip_info.ip));
    ESP_LOGI("ModemHandler", "Netmask     : " IPSTR, IP2STR(&event->ip_info.netmask));
    ESP_LOGI("ModemHandler", "Gateway     : " IPSTR, IP2STR(&event->ip_info.gw));

    ipAddress = "BONKERS"; //IP2STR(&event->ip_info.ip);
    isConnected = true;

    xEventGroupSetBits(ApplicationEvents, COMMS_AVAILABLE_BIT);
    //PppConnectedEvent.notify(true);
  } 
  else if (event_id == IP_EVENT_PPP_LOST_IP)
  {
    ESP_LOGI("ModemHandler", "PPP Connection has been lost.");
```

and all seems to work now.