espressif/esp-matter #864: TC-RR-1.1 Certification issues

eme-bennytang, 2024-03-11:
**Describe the bug**
We have encountered several different issues in certification related to this test case.
Just gathering them here for anyone interested (which I have seen several different user has reported related issue separately)

**1. Sometimes the chip event queue is full and failed to post event**
- Please find this link for more detail logs (https://github.com/project-chip/connectedhomeip/issues/32481)
- Found solution from [jonsmirl](https://github.com/jonsmirl) fixing it by increasing the size of event queue to 50 (https://github.com/espressif/esp-matter/issues/313)
- Not sure why the default size is 25, and we didn't have this problem in another project with earlier stack version (548ef58).
- Anything related to this to share? Otherwise we will just increase it to 50 since we got a lot of static DRAM unused at the moment.

**2. Node label 32 bytes constraint error**
- There is a new issue these recent days and espressif guys have confirmed there is a fix. (d481fa9)
- Referenced issue: https://github.com/espressif/esp-matter/issues/862
- We will implement the fix to see if the same error occurs.

**3. Group Key management command error**
- Found a suggested fix (https://github.com/espressif/esp-matter/issues/765) to increase the "nvs" partition to 48K size
- The problem is, we already increased it to 48K since the beginning of our project and we are still seeing this issue
- Please find this issue for more detail logs (https://github.com/project-chip/connectedhomeip/issues/32481) 
- Is there any further suggested fix for this?  or more information?

**Environment**
- ESP-Matter Commit Id: 2f692ac1
- ESP-IDF Commit Id: v5.1.2
- SoC (eg: ESP32 or ESP32-C3): ESP32S3
- Host Machine OS: Linux
...

wqx6, 2024-03-27:
> Sometimes the chip event queue is full and failed to post event

Matter will post all the UDP messages with specific destination port numbers(5540- Default port for Matter protocol, 5353- Port for mDNS) to Matter task. Each packet will consume an entry in chip event queue. So if your devices are in an environment with lots of mDNS broadcast packets(even the packets are not related to Matter), the event queue will be used up very quickly. This [PR](https://github.com/project-chip/connectedhomeip/pull/31440) might help.

> Group Key management command error

Looks like the error code is https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/group-key-mgmt-server/group-key-mgmt-server.cpp#L528 . Could you add some log to print the error(or change the LogDetail to LogError in the following codes) so that we can know what happened there?

dhrishi, 2024-04-24:
@eme-bennytang Can you please provide the requested details? Or close the issue if resolved

eme-bennytang, 2024-04-25:
@dhrishi  Sorry for late response.
I do figure out the problem a while ago, we have split the nvs partition with another partition for factory data, and this partition is much smaller than the nvs partition (48k).

Once I move the KVS and COUNTERS name space back to nvs as below and it solved the issue.
[image]

Thanks for the reply anyway 
I will close the issue

closed 2024-04-25 by eme-bennytang