espressif/esp-matter #865: Generic_switch add border router functionality

FHFS, 2024-03-11:
I am trying to get the generic_switch example working together with border router.

I have a ESP Thread Border Router, consisting of a esp32s3 and esp32h2.

I got as far as getting the border router enabled and accessible through the CLI.

However I am unable to complete commissioning.

I think it has something to do with mdns. When I start the commissioning proces, wifi gets provisioned and I get some errors like: `E (242238) chip[DIS]: Failed to advertise operational node: b` 

**Does anyone know why I am getting this error when trying to advertise using mdns?**

Full logs down below:

```
I (154) esp_image: segment 1: paddr=0005a848 vaddr=3fc9a8ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037e23e
0x4037e23e: esp_cpu_wait_for_intr at /home/fhfs/Repos/MatterESP/esp-idf/components/esp_hw_support/cpu.c:121

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1758
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbe4
load:0x403cc700,len:0x2d18
entry 0x403c9908
I (26) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (27) boot: compile time Mar 11 2024 10:49:43
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 002ee000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c160020 size=4a820h (305184) map
I (154) esp_image: segment 1: paddr=0005a848 vaddr=3fc9a800 size=05190h ( 20880) load
I (159) esp_image: segment 2: paddr=0005f9e0 vaddr=40374000 size=00638h (  1592) load
I (161) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=15d100h (1429760) map
I (425) esp_image: segment 4: paddr=001bd128 vaddr=40374638 size=160d0h ( 90320) load
I (446) esp_image: segment 5: paddr=001d3200 vaddr=600fe010 size=00004h (     4) load
I (455) boot: Loaded app from partition at offset 0x10000
I (456) boot: Disabling RNG early entropy source...
I (457) cpu_start: Multicore app
I (460) cpu_start: Pro cpu up.
I (464) cpu_start: Starting app cpu, entry point is 0x40375640
0x40375640: call_start_cpu1 at /home/fhfs/Repos/MatterESP/esp-idf/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (479) cpu_start: Pro cpu start user code
I (479) cpu_start: cpu freq: 160000000 Hz
I (479) cpu_start: Application information:
I (479) cpu_start: Project name:     generic_switch
I (480) cpu_start: App version:      1.0
I (480) cpu_start: Compile time:     Mar 11 2024 12:08:19
I (480) cpu_start: ELF file SHA256:  1a8611227d63553c...
I (480) cpu_start: ESP-IDF:          v5.1.2-dirty
I (481) cpu_start: Min chip rev:     v0.0
I (481) cpu_start: Max chip rev:     v0.99 
I (481) cpu_start: Chip rev:         v0.1
I (481) heap_init: Initializing. RAM available for dynamic allocation:
I (482) heap_init: At 3FCB8CA0 len 00030A70 (194 KiB): DRAM
I (482) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
[... 523 lines cut ...]
I (351888) chip[DIS]: Advertise operational node E8A947DEC51E4DC0-0000000004DA5033
E (351888) chip[DIS]: Failed to advertise operational node: b
I (351888) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1 cp=0
E (351898) chip[ZCL]: OpCreds: Got FailSafeTimerExpired
E (351898) chip[ZCL]: OpCreds: Proceeding to FailSafeCleanup on fail-safe expiry!
E (351898) chip[FP]: Reverting pending fabric data for fabric 0x1
E (351908) chip[FP]: Warning: metadata not found during delete of fabric 0x1
I (351948) chip[FP]: Fabric (0x1) deleted.
I (351948) chip[ZCL]: UserLabel: Last Fabric index 0x1 was removed
I (351948) chip[ZCL]: OpCreds: Fabric index 0x1 was removed
I (351948) chip[DIS]: Updating services using commissioning mode 1
I (351948) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1 cp=0
I (352008) chip[TS]: Pending Last Known Good Time: 2024-03-11T11:33:31
I (352018) chip[TS]: Previous Last Known Good Time: 2023-10-14T01:16:48
I (352018) chip[TS]: Reverted Last Known Good Time to previous value
E (352018) chip[FP]: Warning: metadata not found during delete of fabric 0x1
E (352028) chip[ZCL]: OpCreds: failed to delete fabric at index 1: d8
I (352028) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 0 **********
I (352028) app_main: Commissioning failed, fail safe timer expired
I (352038) chip[FS]: Fail-safe cleanly disarmed
E (352038) chip[DL]: ble_gap_adv_set_data failed: Error ESP32:0x05B0001E 3840
E (352038) chip[DL]: Configure Adv Data failed: Error ESP32:0x05B0001E
E (352038) chip[DL]: Disabling CHIPoBLE service due to error: Error ESP32:0x05B0001E
I (352038) app_main: Commissioning window opened
```

wqx6, 2024-03-12:
`b` means `CHIP_ERROR_NO_MEMORY`. We have already found that there was an [issue](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/WiFiDnssdImpl.cpp#L171) which could return `CHIP_ERROR_NO_MEMORY` and it will be fixed with this [change](https://github.com/project-chip/connectedhomeip/pull/32428/files#diff-26af88e6d156424063f0814aa5ebdddda1eb59e4add9ff33ec5935e0e0229407L168-R168). Could you pick the change and try again?

FHFS, 2024-03-12:
@wqx6 Thanks, I will try it now and get back to you

FHFS, 2024-03-12:
Thanks again @wqx6. Amazing, that worked!

closed 2024-03-12 by FHFS