espressif/esp-matter #866: VSCode IDF extension - compatible with esp-matter's idf?

oidebrett, 2024-03-11:
**Describe the question/query that you have**
There was some conflict with my existing installation of esp-idf (installed through VSCode) and matter. I had to disable my extension, uninstall my existing setup and work from the command line (which is not so bad but most people I know use the VSCode extension).
(ii) Strange errors: (sometimes solved by rm -fr /tmp/chip_*)
(a) head of shell variables not a function context
(b) command not found : pw

I had no issue after I remove the VScode idf extension and followed the command line approach as set out in this repo. 

Is there any recommendation on using the VSCode extension? or is it best to avoid this for now and use the command line?

sayondeep, 2024-03-12:
@oidebrett You can use the VS Code extension for matter development. 
For Matter Development you can refer [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETTINGS.md) to configure the extension.

When installing matter please do not clone the repo in tools directory i.e., `/home/user/.espressif` , it is known to cause some issues.

> command not found : pw
This happens when the installation of esp-matter is unsuccessful.

Please add a comment if face any issue during installation or while using the extension.

dhrishi, 2024-03-18:
@oidebrett Please close the issue if your problem is resolved.

oidebrett, 2024-03-18:
Closed. Information provided above

closed 2024-03-18 by oidebrett