espressif/esp-matter #869: generic_switch border router - Thread credentials

FHFS, 2024-03-12:
Hi there,

I've got the generic_switch project running on an ESP Thread Border Router board.
 I've added Border Router functionality, and I can use the matter cli to configure OpenThread on the esp32h2.
Commissioning works thanks to @wqx6.

Now I would like to get Border Router to also get the OpenThread Credentials during commissioning.
How do I achieve that?

Does anyone know why the commissioning process only provisions wifi credentials and not thread credentials?

FHFS, 2024-03-14:
When I enable "CONFIG_ENABLE_MATTER_OVER_THREAD" commissioning fails, see log below.

When I look at `esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/network-commissioning/network-commissioning.cpp` it looks like `CHIP_DEVICE_CONFIG_ENABLE_THREAD` is required for passing `SupportedThreadFeatures` during the commissioning process.

However setting `CHIP_DEVICE_CONFIG_ENABLE_THREAD` through `CONFIG_ENABLE_MATTER_OVER_THREAD` will cause `err = esp_matter::start(app_event_cb);` to try and initialize OpenThread. But for the Border Router we need to initialize OpenThread with `esp_matter::thread_br_init(&config);`. We cannot initialize `esp_matter::thread_br_init(&config);` because that will try to attach to a netif, and we do not have network access yet.

**Is it just not possible yet to get the Thread credentials for the Border Router?**

```
I (163) esp_image: segment 1: paddr=00066124 vaddr=3fc9a8I (516) main_task: Calling app_main()
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40384ac1
0x40384ac1: spimem_flash_ll_cmd_is_done at /home/fhfs/Repos/MatterESP/esp-idf/components/hal/esp32s3/include/hal/spimem_flash_ll.h:60 (discriminator 1)
 (inlined by) spi_flash_hal_poll_cmd_done at /home/fhfs/Repos/MatterESP/esp-idf/components/hal/spi_flash_hal_common.inc:38 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1758
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbe4
load:0x403cc700,len:0x2d18
entry 0x403c9908
I (26) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (27) boot: compile time Mar 14 2024 14:11:38
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 002ee000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c180020 size=560fch (352508) map
I (163) esp_image: segment 1: paddr=00066124 vaddr=3fc9a800 size=051f0h ( 20976) load
I (168) esp_image: segment 2: paddr=0006b31c vaddr=40374000 size=04cfch ( 19708) load
I (173) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=171a9ch (1514140) map
I (449) esp_image: segment 4: paddr=001e1ac4 vaddr=40378cfc size=11a88h ( 72328) load
I (465) esp_image: segment 5: paddr=001f3554 vaddr=600fe010 size=00004h (     4) load
I (475) boot: Loaded app from partition at offset 0x10000
I (475) boot: Disabling RNG early entropy source...
I (476) cpu_start: Multicore app
I (480) cpu_start: Pro cpu up.
I (483) cpu_start: Starting app cpu, entry point is 0x40375644
0x40375644: call_start_cpu1 at /home/fhfs/Repos/MatterESP/esp-idf/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (499) cpu_start: Pro cpu start user code
I (499) cpu_start: cpu freq: 160000000 Hz
I (499) cpu_start: Application information:
I (499) cpu_start: Project name:     generic_switch
I (499) cpu_start: App version:      1.0
I (499) cpu_start: Compile time:     Mar 14 2024 14:11:27
I (500) cpu_start: ELF file SHA256:  03df04ec581b1b2a...
I (500) cpu_start: ESP-IDF:          v5.1.2-dirty
I (500) cpu_start: Min chip rev:     v0.0
I (500) cpu_start: Max chip rev:     v0.99 
I (501) cpu_start: Chip rev:         v0.1
I (501) heap_init: Initializing. RAM available for dynamic allocation:
I (501) heap_init: At 3FCBC378 len 0002D398 (180 KiB): DRAM
I (501) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (502) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (502) heap_init: At 600FE014 len 00001FD4 (7 KiB): RTCRAM
I (503) spi_flash: detected chip: generic
I (503) spi_flash: flash io: dio
[... 632 lines cut ...]
E (120727) OPENTHREAD: esp_openthread_lock_release(30): Failed to release the lock because the mutex is not ready
E (120727) OPENTHREAD: esp_openthread_lock_acquire(21): Failed to acquire the lock because the mutex is not ready
E (120727) OPENTHREAD: esp_openthread_lock_release(30): Failed to release the lock because the mutex is not ready
E (130737) OPENTHREAD: esp_openthread_lock_acquire(21): Failed to acquire the lock because the mutex is not ready
E (130737) OPENTHREAD: esp_openthread_lock_release(30): Failed to release the lock because the mutex is not ready
E (130737) OPENTHREAD: esp_openthread_lock_acquire(21): Failed to acquire the lock because the mutex is not ready
E (130737) OPENTHREAD: esp_openthread_lock_release(30): Failed to release the lock because the mutex is not ready
E (140747) OPENTHREAD: esp_openthread_lock_acquire(21): Failed to acquire the lock because the mutex is not ready
E (140747) OPENTHREAD: esp_openthread_lock_release(30): Failed to release the lock because the mutex is not ready
E (140747) OPENTHREAD: esp_openthread_lock_acquire(21): Failed to acquire the lock because the mutex is not ready
E (140747) OPENTHREAD: esp_openthread_lock_release(30): Failed to release the lock because the mutex is not ready
I (142067) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)
I (142067) chip[DL]: Device already advertising, stop active advertisement and restart
[... cut ...]
I (142067) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (142067) NimBLE: 

E (150757) OPENTHREAD: esp_openthread_lock_acquire(21): Failed to acquire the lock because the mutex is not ready
E (150757) OPENTHREAD: esp_openthread_lock_release(30): Failed to release the lock because the mutex is not ready
E (150757) OPENTHREAD: esp_openthread_lock_acquire(21): Failed to acquire the lock because the mutex is not ready
E (150757) OPENTHREAD: esp_openthread_lock_release(30): Failed to release the lock because the mutex is not ready
```

jonsmirl, 2024-03-15:
I'm not a thread user, but I believe this is being addressed in Matter 1.3.

FHFS, 2024-03-15:
@jonsmirl Thanks for your response!
Do you by any chance know if there is a upstream issue about the problem I am having? Or why do you know it is being addressed in 1.3?

jonsmirl, 2024-03-15:
My understanding is that thread initially got built as three islands. For things commissioned on Google, you had to use a Google bridge, things on Apple an Apple bridge, etc. Thread was not shared between the ecosystems. 1,3 fixes it to work as a single system.

FHFS, 2024-03-15:
Hmmm, but I am not trying to mix Google and Apple.
I am trying to commission a generic_switch, which also has border router capabilities.

Does anyone know what happens with the esp32c6 that has wifi and thread capabilities?
Can the esp32c6 get the wifi and thread credentials during commissioning?

jonsmirl, 2024-03-16:
This is what is coming in Matter 1.3
https://www.threadgroup.org/news-events/blog/ID/428/Threads-2024-Enhancements-and-What-Theyll-Mean-to-You
https://www.theverge.com/2024/1/8/24028203/thread-group-fix-credential-sharing-thread-border-router

wqx6, 2024-03-18:
Hi, @FHFS Matter Working Group is working on a new device type named Home Router and Access Point (HRAP) and you can use related clusters to configure the Thread Dataset of the Thread Border Router. The device type might be planed to be published in the end of year.

For now, the Thread Border can be commissioned as a Matter-Over-Wi-Fi device and the Thread credential can be configured after commissioning with some custom cluster. Just like what we do in the rainmaker cluster. https://github.com/espressif/esp-matter/blob/main/components/esp_matter_rainmaker/esp_matter_rainmaker.cpp#L362

FHFS, 2024-03-18:
@wqx6 Thanks again! HRAP sounds exactly like what I am looking for.

> For now, the Thread Border can be commissioned as a Matter-Over-Wi-Fi device and the Thread credential can be configured after commissioning with some custom cluster. Just like what we do in the [...]

What you are suggesting here is, I should configure Thread after Commissioning over Wifi, using a custom cluster. But how do I get the Thread Credentials in this case? In rainmaker it is probably done using the android/iOS app?

wqx6, 2024-03-20:
Current iOS has provided APIs to manage thread credentials https://developer.apple.com/documentation/threadnetwork/managing-thread-network-credentials , after the BR is added to your fabric you can get the credentials with these APIs and set them with custom cluster client on your phone app. We have already tested these APIs in our test iOS app.

For Android, I think there should be similar APIs.

FHFS, 2024-03-20:
Yes I did see the ThreadCredentials being requested  in both the iOS and Android Rainmaker sources.

Thank you both for your help and input!
I look forward to seeing the HRAP in the next Matter version.

closed 2024-03-20 by FHFS