espressif/esp-matter #871: Connect matter device to Google Home won't work

FloBer431, 2024-03-13:
Hello there,

I finally got my contact-sensor device generally working.
I also already connected it with apple home and it working out fine (except that the default state is open, not closed as i want it to be)
But when I tried to connect it with Google Home, following error occured:
`E (45302) chip[DMG]: Error retrieving data from clusterId: 0x0000_0028, err = 201`

maybe somebody can help

Thanks
Florian

VaishaliAvhale, 2024-03-18:
Please enable debug log as follows Component config → Log output → Default log verbosity -> Debug
and retry same thing and attach complete device side logs file.

FloBer431, 2024-03-19:
I enabled the debug mode and here is the complete output:

`I (36161) chip[ZCL]: Received command to open commissioning window
I (36161) chip[DIS]: Updating services using commissioning mode 2
I (36171) chip[DIS]: CHIP minimal mDNS started advertising.
I (36181) chip[DIS]: Advertise operational node 5236AFA001B879D3-00000000B69FF720
I (36181) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 5236AFA001B879D3-00000000B69FF720.
I (36201) chip[DIS]: mDNS service published: _matter._tcp
I (36201) chip[DIS]: Advertise commission parameter vendorID=65522 productID=32769 discriminator=4054/15 cm=2 cp=0
I (36211) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 7E0318A45F6ECB60.
I (36231) chip[DIS]: mDNS service published: _matterc._udp
I (36231) chip[ZCL]: Commissioning window is now open
I (36241) chip[EM]: <<< [E:17196r S:58560 M:121875092 (Ack:140751351)] (S) Msg TX to 1:00000000C2053460 [79D3] [UDP:192.168.178.109%st1:57784] --- Type 0001:09 (IM:InvokeCommandResponse)
I (36261) logger: [CSystem::matter_event_callback] Commissioning window opened [system.cpp:382]
I (36261) chip[EM]: >>> [E:17196r S:58560 M:140751352 (Ack:121875092)] (S) Msg RX from 1:00000000C2053460 [79D3] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (36411) chip[EM]: >>> [E:53360r S:0 M:214017046] (U) Msg RX from 0:8E41A27935CD5049 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
I (36421) chip[EM]: <<< [E:53360r S:0 M:147602092 (Ack:214017046)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (36441) chip[EM]: <<< [E:53360r S:0 M:147602093 (Ack:214017046)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0000:21 (SecureChannel:PBKDFParamResponse)
I (36451) chip[SVR]: Commissioning session establishment step started
I (36481) chip[EM]: >>> [E:53360r S:0 M:214017047 (Ack:147602093)] (U) Msg RX from 0:8E41A27935CD5049 [0000] --- Type 0000:22 (SecureChannel:PASE_Pake1)
I (36481) chip[EM]: <<< [E:53360r S:0 M:147602094 (Ack:214017047)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (37231) chip[EM]: <<< [E:53360r S:0 M:147602095 (Ack:214017047)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0000:23 (SecureChannel:PASE_Pake2)
E (37241) chip[DL]: Long dispatch time: 763 ms, for event type 3
I (37271) chip[EM]: >>> [E:53360r S:0 M:214017048 (Ack:147602095)] (U) Msg RX from 0:8E41A27935CD5049 [0000] --- Type 0000:24 (SecureChannel:PASE_Pake3)
I (37271) chip[EM]: <<< [E:53360r S:0 M:147602096 (Ack:214017048)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (37291) chip[EM]: <<< [E:53360r S:0 M:147602097 (Ack:214017048)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0000:40 (SecureChannel:StatusReport)
I (37311) chip[SC]: SecureSession[0x3fc9fe78, LSID:58562]: State change 'kEstablishing' --> 'kActive'
I (37321) chip[SVR]: Commissioning completed session establishment step
I (37321) chip[DIS]: Updating services using commissioning mode 0
I (37331) chip[DIS]: CHIP minimal mDNS started advertising.
I (37351) chip[DIS]: Advertise operational node 5236AFA001B879D3-00000000B69FF720
I (37361) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 5236AFA001B879D3-00000000B69FF720.
I (37371) chip[DIS]: mDNS service published: _matter._tcp
I (37371) chip[SVR]: Device completed Rendezvous process
I (37381) logger: [CSystem::matter_event_callback] Commissioning session started [system.cpp:374]
I (37391) chip[EM]: >>> [E:53360r S:0 M:214017049 (Ack:147602097)] (U) Msg RX from 0:8E41A27935CD5049 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (37401) chip[EM]: >>> [E:53361r S:58562 M:14345287] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
I (37411) logger: [CSystem::matter_event_callback] Commissioning window closed [system.cpp:385]
I (37421) chip[EM]: <<< [E:53361r S:58562 M:267819773 (Ack:14345287)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0001:05 (IM:ReportData)
I (37481) chip[EM]: >>> [E:53361r S:58562 M:14345288 (Ack:267819773)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (37481) chip[EM]: >>> [E:53362r S:58562 M:14345289] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (37501) esp_matter_command: Received command 0x00000000 for endpoint 0x0000's cluster 0x00000030
I (37511) chip[FS]: GeneralCommissioning: Received ArmFailSafe (120s)
I (37511) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 0 **********
I (37531) chip[EM]: <<< [E:53362r S:58562 M:267819774 (Ack:14345289)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0001:09 (IM:InvokeCommandResponse)
I (37551) chip[EM]: >>> [E:53362r S:58562 M:14345290 (Ack:267819774)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (38261) ROUTE_HOOK: Received RIO
I (38261) ROUTE_HOOK: prefix FDD9:AD53:1ECD:: lifetime 1800
[... cut ...]
E (41771) chip[DMG]: Error retrieving data from clusterId: 0x0000_0028, err = 201
I (41781) chip[EM]: <<< [E:53374r S:58562 M:267819787 (Ack:14345313)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0001:05 (IM:ReportData)
I (41821) chip[EM]: >>> [E:53374r S:58562 M:14345314 (Ack:267819787)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (42751) ROUTE_HOOK: Received RIO
I (42751) ROUTE_HOOK: prefix FDD9:AD53:1ECD:: lifetime 1800
I (46951) ROUTE_HOOK: Received RIO`

FHFS, 2024-03-19:
I added some breaklines for readability, because I couldnt read this.
```
I (36161) chip[ZCL]: Received command to open commissioning window 
I (36161) chip[DIS]: Updating services using commissioning mode 2 
I (36171) chip[DIS]: CHIP minimal mDNS started advertising. 
I (36181) chip[DIS]: Advertise operational node 5236AFA001B879D3-00000000B69FF720 
I (36181) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 5236AFA001B879D3-00000000B69FF720. 
I (36201) chip[DIS]: mDNS service published: _matter._tcp 
I (36201) chip[DIS]: Advertise commission parameter vendorID=65522 productID=32769 discriminator=4054/15 cm=2 cp=0 
I (36211) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 7E0318A45F6ECB60. 
I (36231) chip[DIS]: mDNS service published: _matterc._udp 
I (36231) chip[ZCL]: Commissioning window is now open 
I (36241) chip[EM]: <<< [E:17196r S:58560 M:121875092 (Ack:140751351)] (S) Msg TX to 1:00000000C2053460 [79D3] [UDP:192.168.178.109%st1:57784] --- Type 0001:09 (IM:InvokeCommandResponse) 
I (36261) logger: [CSystem::matter_event_callback] Commissioning window opened [system.cpp:382] 
I (36261) chip[EM]: >>> [E:17196r S:58560 M:140751352 (Ack:121875092)] (S) Msg RX from 1:00000000C2053460 [79D3] --- Type 0000:10 (SecureChannel:StandaloneAck) 
I (36411) chip[EM]: >>> [E:53360r S:0 M:214017046] (U) Msg RX from 0:8E41A27935CD5049 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest) 
I (36421) chip[EM]: <<< [E:53360r S:0 M:147602092 (Ack:214017046)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0000:10 (SecureChannel:StandaloneAck) 
I (36441) chip[EM]: <<< [E:53360r S:0 M:147602093 (Ack:214017046)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0000:21 (SecureChannel:PBKDFParamResponse) 
I (36451) chip[SVR]: Commissioning session establishment step started 
I (36481) chip[EM]: >>> [E:53360r S:0 M:214017047 (Ack:147602093)] (U) Msg RX from 0:8E41A27935CD5049 [0000] --- Type 0000:22 (SecureChannel:PASE_Pake1) 
I (36481) chip[EM]: <<< [E:53360r S:0 M:147602094 (Ack:214017047)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0000:10 (SecureChannel:StandaloneAck) 
I (37231) chip[EM]: <<< [E:53360r S:0 M:147602095 (Ack:214017047)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0000:23 (SecureChannel:PASE_Pake2) 
E (37241) chip[DL]: Long dispatch time: 763 ms, for event type 3 
I (37271) chip[EM]: >>> [E:53360r S:0 M:214017048 (Ack:147602095)] (U) Msg RX from 0:8E41A27935CD5049 [0000] --- Type 0000:24 (SecureChannel:PASE_Pake3) 
I (37271) chip[EM]: <<< [E:53360r S:0 M:147602096 (Ack:214017048)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0000:10 (SecureChannel:StandaloneAck) 
I (37291) chip[EM]: <<< [E:53360r S:0 M:147602097 (Ack:214017048)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0000:40 (SecureChannel:StatusReport) 
I (37311) chip[SC]: SecureSession[0x3fc9fe78, LSID:58562]: State change 'kEstablishing' --> 'kActive' 
I (37321) chip[SVR]: Commissioning completed session establishment step 
I (37321) chip[DIS]: Updating services using commissioning mode 0 
I (37331) chip[DIS]: CHIP minimal mDNS started advertising. 
I (37351) chip[DIS]: Advertise operational node 5236AFA001B879D3-00000000B69FF720 
I (37361) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 5236AFA001B879D3-00000000B69FF720. 
I (37371) chip[DIS]: mDNS service published: _matter._tcp 
I (37371) chip[SVR]: Device completed Rendezvous process 
I (37381) logger: [CSystem::matter_event_callback] Commissioning session started [system.cpp:374] 
I (37391) chip[EM]: >>> [E:53360r S:0 M:214017049 (Ack:147602097)] (U) Msg RX from 0:8E41A27935CD5049 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck) 
I (37401) chip[EM]: >>> [E:53361r S:58562 M:14345287] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest) 
I (37411) logger: [CSystem::matter_event_callback] Commissioning window closed [system.cpp:385] 
I (37421) chip[EM]: <<< [E:53361r S:58562 M:267819773 (Ack:14345287)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0001:05 (IM:ReportData) 
I (37481) chip[EM]: >>> [E:53361r S:58562 M:14345288 (Ack:267819773)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck) 
I (37481) chip[EM]: >>> [E:53362r S:58562 M:14345289] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest) 
I (37501) esp_matter_command: Received command 0x00000000 for endpoint 0x0000's cluster 0x00000030 
I (37511) chip[FS]: GeneralCommissioning: Received ArmFailSaf
E (120s) 
I (37511) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 0 ********** 
I (37531) chip[EM]: <<< [E:53362r S:58562 M:267819774 (Ack:14345289)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0001:09 (IM:InvokeCommandResponse) 
I (37551) chip[EM]: >>> [E:53362r S:58562 M:14345290 (Ack:267819774)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck) 
I (38261) ROUTE_HOOK: Received RIO 
I (38261) ROUTE_HOOK: prefix FDD9:AD53:1ECD:: lifetime 1800 
[... cut ...]
E (41771) chip[DMG]: Error retrieving data from clusterId: 0x0000_0028, err = 201 
I (41781) chip[EM]: <<< [E:53374r S:58562 M:267819787 (Ack:14345313)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:62859] --- Type 0001:05 (IM:ReportData) 
I (41821) chip[EM]: >>> [E:53374r S:58562 M:14345314 (Ack:267819787)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck) 
I (42751) ROUTE_HOOK: Received RIO 
I (42751) ROUTE_HOOK: prefix FDD9:AD53:1ECD:: lifetime 1800 
I (46951) ROUTE_HOOK: Received RIO
```

VaishaliAvhale, 2024-03-21:
The above logs do not indicate that debug log configuration is enabled.
The error 201 stands for CHIP_DEVICE_ERROR_CONFIG_NOT_FOUND, indicating that the requested configuration value was not found. The error is encountered for cluster 0x0028, which is the basic information cluster. Please enable the debug log as mentioned above to get the attribute of the cluster for which the config value is not found.
Additionally, could you please let me know if you are trying to make any changes to the basic information cluster in your code?

FloBer431, 2024-03-21:
@VaishaliAvhale 
I tried to enable the debug log as mentioned, but I think it didn't work. I'm working with Visual Studio on Windows via WSL. I enabled the debug log in the VS Extension settings, but this dont effect my log output.
Is there another way to enable the log?

I don't think I'm trying to do any changes to the basic information cluster.

FHFS, 2024-03-21:
Add `CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y` to your file `sdkconfig.defaults`
And build using `idf.py -D SDKCONFIG_DEFAULTS="sdkconfig.defaults" clean build`

You should see `CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y` in the `sdkconfig` that was generated during the build.

FloBer431, 2024-03-21:
Okay, I think that worked now, thank you @FHFS.

This is the part from the `sdkconfig`:
```
# Log output
#
# CONFIG_LOG_DEFAULT_LEVEL_NONE is not set
# CONFIG_LOG_DEFAULT_LEVEL_ERROR is not set
# CONFIG_LOG_DEFAULT_LEVEL_WARN is not set
# CONFIG_LOG_DEFAULT_LEVEL_INFO is not set
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y
# CONFIG_LOG_DEFAULT_LEVEL_VERBOSE is not set
```

Now the log is massive, I hope I copied everything that is necessary:

```
D (63516) chip[DMG]: All ReadHandler-s are clean, clear GlobalDirtySet
I (63526) chip[EM]: >>> [E:30269r S:55454 M:225728011 (Ack:190411351)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
D (63536) chip[EM]: Found matching exchange: 30269r, Delegate: 0x0
D (63546) chip[EM]: Rxd Ack; Removing MessageCounter:190411351 from Retrans Table on exchange 30269r
I (63556) chip[EM]: >>> [E:30270r S:55454 M:225728012] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
D (63566) chip[EM]: Handling via exchange: 30270r, Delegate: 0x3fca5054
D (63576) chip[IM]: Received Read request
D (63576) chip[DMG]: ReadRequestMessage =
D (63586) chip[DMG]: {
D (63586) chip[DMG]:    AttributePathIBs =
D (63586) chip[DMG]:    [
D (63596) chip[DMG]:            AttributePathIB =
D (63596) chip[DMG]:            {
D (63596) chip[DMG]:                    Endpoint = 0x0,
D (63606) chip[DMG]:                    Cluster = 0x28,
D (63606) chip[DMG]:                    Attribute = 0x0000_0003,
D (63606) chip[DMG]:            }
D (63616) chip[DMG]: 
D (63616) chip[DMG]:    ],
D (63616) chip[DMG]: 
D (63626) chip[DMG]:    isFabricFiltered = true, 
D (63626) chip[DMG]:    InteractionModelRevision = 11
D (63626) chip[DMG]: },
D (63636) chip[DMG]: IM RH moving to [CanStartReporting]
D (63636) chip[DMG]: Building Reports for ReadHandler with LastReportGeneration = 0x0000000000000000 DirtyGeneration = 0x0000000000000000
D (63646) chip[DMG]: <RE:Run> Cluster 28, Attribute 3 is dirty
D (63656) chip[DMG]: Reading attribute: Cluster=0x0000_0028 Endpoint=0 AttributeId=0x0000_0003 (expanded=0)
D (63666) nvs: nvs_open_from_partition chip-factory 0
D (63676) nvs: nvs_get_str_or_blob product-name
D (63676) nvs: nvs_close 381
E (63676) chip[DMG]: Error retrieving data from clusterId: 0x0000_0028, err = 201
D (63686) chip[DMG]: <RE> Sending report (payload has 33 bytes)...
I (63696) chip[EM]: <<< [E:30270r S:55454 M:190411352 (Ack:225728012)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:59913] --- Type 0001:05 (IM:ReportData)
D (63716) chip[DMG]: <RE> OnReportConfirm: NumReports = 0
D (63716) chip[DMG]: <RE> ReportsInFlight = 0 with readHandler 0, RE has no more messages
D (63726) chip[DMG]: IM RH moving to [AwaitingDestruction]
D (63726) nvs: nvs_open_from_partition CHIP_KVS 0
D (63736) nvs: nvs_get_str_or_blob g/su/0
D (63736) nvs: nvs_close 382
D (63746) nvs: nvs_open_from_partition CHIP_KVS 0
D (63746) nvs: nvs_get_str_or_blob g/su/1
D (63746) nvs: nvs_close 383
D (63756) nvs: nvs_open_from_partition CHIP_KVS 0
D (63756) nvs: nvs_get_str_or_blob g/su/2
D (63766) nvs: nvs_close 384
D (63766) nvs: nvs_open_from_partition CHIP_KVS 0
D (63766) nvs: nvs_get_str_or_blob g/su/3
D (63776) nvs: nvs_close 385
D (63776) nvs: nvs_open_from_partition CHIP_KVS 0
D (63786) nvs: nvs_get_str_or_blob g/su/4
D (63786) nvs: nvs_close 386
D (63786) nvs: nvs_open_from_partition CHIP_KVS 0
D (63796) nvs: nvs_get_str_or_blob g/su/5
[... 258 lines cut ...]
D (65916) nvs: nvs_get_str_or_blob g/su/b
D (65916) nvs: nvs_close 441
D (65916) nvs: nvs_open_from_partition CHIP_KVS 0
D (65926) nvs: nvs_get_str_or_blob g/su/c
D (65926) nvs: nvs_close 442
D (65936) nvs: nvs_open_from_partition CHIP_KVS 0
D (65936) nvs: nvs_get_str_or_blob g/su/d
D (65936) nvs: nvs_close 443
D (65946) nvs: nvs_open_from_partition CHIP_KVS 0
D (65946) nvs: nvs_get_str_or_blob g/su/e
D (65956) nvs: nvs_close 444
D (65956) chip[DMG]: All ReadHandler-s are clean, clear GlobalDirtySet
D (65966) chip[IN]: Received a duplicate message with MessageCounter:225728018 on exchange 30273r
I (65966) chip[EM]: >>> [E:30273r S:55454 M:225728018] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
D (65986) chip[EM]: Found matching exchange: 30273r, Delegate: 0x0
D (65986) chip[EM]: Forcing tx of solitary ack for duplicate MessageCounter:225728018 on exchange 30273r
I (65996) chip[EM]: <<< [E:30273r S:55454 M:190411356 (Ack:225728018)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[FE80::A2:2C39:3B2C:AD67%st1]:59913] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (66016) chip[EM]: >>> [E:30273r S:55454 M:225728019 (Ack:190411355)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
D (66036) chip[EM]: Found matching exchange: 30273r, Delegate: 0x0
D (66036) chip[EM]: Rxd Ack; Removing MessageCounter:190411355 from Retrans Table on exchange 30273r
D (74896) wifi:obss scan is disabled
D (74896) wifi:start obss scan: obss scan is stopped
E (82256) chip[SWU]: No suitable OTA Provider candidate found
I (82256) chip[SWU]: No provider available
```

FloBer431, 2024-04-03:
Nobody who can help me here?

FHFS, 2024-04-03:
You could check your partition. Your error shows it was trying to get the product name from the nvs partition chip-factory.
Check if there is a valid name there?
```
D (63666) nvs: nvs_open_from_partition chip-factory 0
D (63676) nvs: nvs_get_str_or_blob product-name
D (63676) nvs: nvs_close 381
E (63676) chip[DMG]: Error retrieving data from clusterId: 0x0000_0028, err = 201
```

I do know generic_switch example has some factory data you need to flash before running your application. See [generic_switch](https://github.com/espressif/esp-matter/tree/main/examples/generic_switch#1-additional-environment-setup)

jadhavrohit924, 2024-08-01:
@FloBer431 Are you still facing the issue?

dhrishi, 2024-10-15:
Closing this now. Please re-open if you still see this issue