espressif/esp-matter #873: Air quality measurement issue

internax, 2024-03-16:
Hello, I have problem with setting up the AirQualitySensor device. I want to add CO2 measurement to the node. In the Matter Specification of the device type AirQualityMeasurement it is written that cluster requirements are of course Air Quality cluster, Identification cluster and as well many measurement clusters. These clusters are not in air_quality_sensor in esp-matter. According to [this issue posted previously](https://github.com/espressif/esp-matter/issues/628) I've added new cluster to air_quality_sensor:

`  air_quality_sensor::config_t air_quality_sensor_config;
    endpoint_t *endpoint_air_quality_sensor = air_quality_sensor::create(node, &air_quality_sensor_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint_air_quality_sensor != nullptr, ESP_LOGE(TAG, "Failed to create air quality sensor device endpoint"));

    cluster::create(endpoint_air_quality_sensor, CarbonDioxideConcentrationMeasurement::Id, CLUSTER_FLAG_SERVER);
    cluster_t *cluster_co2 = cluster::get(endpoint_air_quality_sensor, CarbonDioxideConcentrationMeasurement::Id);
    ABORT_APP_ON_FAILURE(cluster_co2 != nullptr, ESP_LOGE(TAG, "Failed to create air quality sensor device cluster"));

    cluster::carbon_dioxide_concentration_measurement::feature::numeric_measurement::config_t feature_co2_config;
    feature_co2_config.measured_value = 1000;
    feature_co2_config.min_measured_value = 1;
    feature_co2_config.max_measured_value = 10000;
    feature_co2_config.uncertainty = 0;
    feature_co2_config.measurement_unit = 0;
    cluster::carbon_dioxide_concentration_measurement::feature::numeric_measurement::add(cluster_co2, &feature_co2_config);`

Cluster is attached to the endpoint, I am able to write value to the cluster, read the written value. My problem is that I can't see CO2 sensor in my apple home (I am able to see other endpoints on the same node). I want to ask for an advice. Am I doing something wrong in my code or I misunderstanded the logic behind this?

Best regards from Czechia!

Pips801, 2024-03-17:
I don't think those kind of matter sensors work in Apple Home, only the HomeKit ones. Nobody has updated their apps to support matter 1.2

lboue, 2024-03-17:
You can try your Matter device with home-assistant. It has support: [Matter sensors for air quality measurements](https://github.com/home-assistant/core/pull/108173).

crealodi, 2024-03-18:
Thank you for raising this issue. I encountered the same issue and I tried commissioning the sensor in HomeAssistant. 
I was able to do so - but the CO2 value is not updated while the temperature values are continuously updated. The CO2 shown is the value present during commissioning.

I tried reading the CO2 measured-value by commissioning the same node with chip-tool. There I get the following error: 
`CHIP:TOO: Response Failure: Can not decode Data
CHIP:TOO: Run command failure: src/lib/core/TLVReader.cpp:283: CHIP Error 0x00000026: Wrong TLV type`

Following this error I tried to change the data types according to https://github.com/YOGYUI/matter-esp32-scd41 but didn't succeed. After finding out, that @YOGYUI fixed this entirely and has an open commit addressing this issue I got it running with his fixed files: https://github.com/YOGYUI/esp-matter/commit/810a9ca8bcbf0dbe1baf36b18d7014e715b2f8f4 - at least in HomeAssistant since - as mentioned before - CO2 seems not to be supported in Apple Home. 

Thank you @YOGYUI for the solution and @internax for raising the issue.

lboue, 2024-03-18:
> Thank you for raising this issue. I encountered the same issue and I tried commissioning the sensor in HomeAssistant. I was able to do so - but the CO2 value is not updated while the temperature values are continuously updated. The CO2 shown is the value present during commissioning.

If it comes from HA, you can report an issue here: [Report an issue with Home Assistant Core](https://github.com/home-assistant/core/issues/new?assignees=&labels=&projects=&template=bug_report.yml)

internax, 2024-03-22:
Thank everyone for your advice. I ll probably pass data to homekit in form of some measurement device and switch to Air quality measuremet device later, when matter 1.2 will be supported in homekit. Do anyone knows when matter 1.2 in homekit will be supported?

closed 2024-03-22 by internax