espressif/esp-matter #874: Turn off Rainmaker

jonsmirl, 2024-03-17:
How do I turn off Rainmaker? I don't see an obvious config option to turn it off.

shubhamdp, 2024-03-18:
@jonsmirl Rainmaker is not enabled in this repository, we have all examples in esp-rainmaker repository https://github.com/espressif/esp-rainmaker/tree/master/examples/matter/.

In this repo, what we have is just the declaration and definitions of the clusters required by rainmaker.

It would be helpful if you could elaborate more on the problem.

jonsmirl, 2024-03-18:
I am using some library code which kept failing on:
~~~
    esp_sntp_setoperatingmode(SNTP_OPMODE_POLL);
    esp_sntp_setservername(0, server);
    esp_sntp_init();
~~~
I thought it might be conflicting with Rainmaker which also uses NTP.
~~~
  if (!esp_sntp_enabled()) {
    esp_sntp_setoperatingmode(SNTP_OPMODE_POLL);
    esp_sntp_setservername(0, server);
    esp_sntp_init();
  }
~~~
If I protect it with esp_sntp_enabled() it doesn't fail. My stuff is working now and I did not track down which other code was turning on NTP.

closed 2024-03-18 by jonsmirl