espressif/esp-matter #875: all_device_types_app console issue

sdrasti, 2024-03-19:
When using the UART0 for console I/O the all_device_types_app works as expected. But when I set the USB/JTAG serial port for the console on a ESP32-C6 Devkit, I get the following output:

```
I (665) main_task: Calling app_main()
I (685) app_main: 

Enter command 'create --device' for a list of available devices
esp32c6> 
esp32c6> 
esp32c6> 
esp32c6> ;R[31R[31;R[31R[31;R[31R
Unrecognized command
esp32c6> R[31;0
Unrecognized command
esp32c6> 
esp32c6> 
esp32c6> ;R[31;0
Unrecognized command
esp32c6> 
esp32c6> 
esp32c6> 
esp32c6> 31;0
Unrecognized command
esp32c6> 
esp32c6> 
esp32c6> 
esp32c6> [31R
Unrecognized command
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
```

When I comment out `example::console::init();` in app_main.cpp but still leave `CONFIG_ENABLE_CHIP_SHELL=y` then I get proper log messages of the Matter stack and a prompt shows up where I can enter Matter and Wifi commands without any issues.

Is there anything that needs to be configured in `static void initialize_console(void)` in esp_matter_console_helpers.cpp to make the console work over the USB/JTAG serial for selecting a device type?

**Environment**
- ESP-Matter Commit Id: 8bd97db
- ESP-IDF Commit Id: v5.1.2
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6 DevkitM-1
- Host Machine OS: MacOS

jonsmirl, 2024-03-19:
Try a different terminal program, these ";R[31R[31;R[31R[31;R[31R" are responses to escape sequence queries from the terminal.

shubhamdp, 2024-03-22:
Did you use the UART or USB port on C6?

If you used the USB port then you need to enable `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG` config option.

sdrasti, 2024-03-25:
@shubhamdp `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG ` is enabled, and as mentioned, the log output and the chip-shell commands work as expected when using the USB port. The issue is related to the console that is added in `esp_matter_console_helpers.cpp`

shubhamdp, 2024-05-06:
I built light with `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y` and I was able to use the console. Is this still an issue, if yes, can you please share the sdkconfig.

dhrishi, 2024-05-15:
@sdrasti Please provide the requested details. Or close the issue if resolved

sdrasti, 2024-05-15:
@shubhamdp @dhrishi 
With the light example app running on a ESP-C6 I got the Matter Console also working along with the logging console without any issues. It seemed to me, as if the the problem was linked to the code in esp_matter_console_helpers.cpp (that adds additional commands to the matter console for choosing a device types) of the all_device_types_app. Anyhow, I haven't tried to track down the issue any further, so I would close this issue then.