espressif/esp-matter #877: Time Format Localization, Localization Config

jonsmirl, 2024-03-19:
~~~
    /* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
    node::config_t node_config;
    node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
    endpoint_t *root = endpoint::get(node, 0);
    time_format_localization::config_t tfl_config;
    time_format_localization::create(root, &tfl_config, CLUSTER_FLAG_SERVER, NULL);
    localization_configuration::config_t locale_config;
    localization_configuration::create(root, &locale_config, CLUSTER_FLAG_SERVER);
~~~
produces these errors
~~~
E (1923) chip[ZCL]: Failed to read calendar type with error: 0x86
I (1923) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000002B's Attribute 0x00000000 is  **********
I (1923) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x0000002B's Attribute 0x00000000 is `��?@��?
                                                                                                                      **********
E (1923) chip[ZCL]: Failed to write active locale with error: 0x87
I (1923) esp_matter_core: Dynamic endpoint 0 added
~~~
Hour format does not appear to be stored in flash.

sayondeep, 2024-03-20:
@jonsmirl you need to generate factory partition using mfg_tool.py with 
--calendar-types and --locales  arguments to specify allowed values for supported calender types and locales only then it can write successfully if the values are among supported calendar types and locales respectively. You can follow the instructions [here](https://github.com/espressif/esp-matter/blob/main/tools/mfg_tool/README.md)

You also need to set Device_info_provider to factory in order to read the values from factory partition. You can see [here](https://github.com/espressif/connectedhomeip/blob/9f6d627e0262e1d023986291948bb4e845be803e/src/app/clusters/time-format-localization-server/time-format-localization-server.cpp#L122) 

> `time_format_localization::create(root, &tfl_config, CLUSTER_FLAG_SERVER, NULL);`

Change the NULL to 1 as it is used as a bitmap for enabling calender_format feature.

jonsmirl, 2024-03-20:
I didn't turn the calendar types feature on.

jonsmirl, 2024-03-21:
This localization cluster design doesn't work when you OTA in different images for different languages. What if we add translated images after the device ships?

wqx6, 2024-05-13:
> That print comes up when initializing the cluster, the cluster init method also try to touch the active-calendar-type and supported-calendar-types attributes even if calendar format feature is not enabled. Fixing that in https://github.com/project-chip/connectedhomeip/pull/32850. 

Fixed in https://github.com/project-chip/connectedhomeip/pull/32747

dhrishi, 2024-05-13:
Looks like the fix is not yet synced to esp-matter connectedhomeip submodule. Will re-open, until this is confirmed

dhrishi, 2024-07-26:
Closing as the fix is now synced

closed 2024-07-26 by dhrishi