espressif/esp-matter #878: Device certification and production requirements for firmware

damian-kurek-wizzdev, 2024-03-22:
My matter device will go into the certification soon and right after it will go into production. I read the matter documentation about this topic and have a few questions. 

- I guess that I can leave test certificates as they are for the certification process. 
- Do I need to add something extra to the partition table?
- How do you change certificates to use one flashed during production?
- Is there anything else I need to do after getting certificates?

wqx6, 2024-03-27:
> I guess that I can leave test certificates as they are for the certification process.

Yes, you can use test certificates for certification test. But please **DO NOT** use the test root PAA certificates in [chip repo](https://github.com/espressif/connectedhomeip/tree/master/credentials/test/attestation) for certification test. You can choose to generate a test PAA cert-key and generate the Device Attestation Chains for the test products from the test PAA cert-key. Or you can use the Device Attestation Chains from your PKI provider. The [document](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/certification.html#certificates-and-keys) might help you

> Do I need to add something extra to the partition table?

If you don't select any other clusters in your PICS files (such as LocalizationConfiguration cluster) which require a `DeviceInfoProvider`, I think nothing extra should be added.

> How do you change certificates to use one flashed during production?

Your PKI provider should be able to help you on that. If you are using our Pre-Provisioning service(https://www.espressif.com/en/support/services#manufacturing), we will send you the modules with DAC chains flashed.

> Is there anything else I need to do after getting certificates?

CSA will send you a CD file and you should replace the CD file in your products. Also you need to make sure that the DAC chains in your products are generated from the PAA certificates in the [DCL MainNET](https://dcl.iotledger.io/pki)

dhrishi, 2024-04-01:
Closing as the questions are answered

closed 2024-04-01 by dhrishi