espressif/esp-matter #879: TC-SU-2.5 OTA is executed immediately

svenehlers, 2024-03-22:
**Describe the bug**
My Test provider faced this issue with the Testcase TC-SU-2.5 and i reproduced it manually:

OTA provider app is set up with this command:

./chip-ota-provider-app --filepath ota.bin -a proceed -p 180

device (node id 1) and ota provider app (node id 2) are commissioned using chip tool interactive mode.

Permission is given by

 accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets": [{"cluster": 41, "endpoint": null, "deviceType": null}]}]' 2 0

ota is announced by

otasoftwareupdaterequester announce-otaprovider 2 0x1494 0 0 1 0

Software Update is executed immediately without 180s delay but it should take 180s before executing.

**Environment**
- ESP-Matter Commit Id: dd4f34e
- ESP-IDF Commit Id: e088c3766b
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file): Device logs are not available but i will provide them with delay if it helps.
- ota provider app logs: [Logs.txt](https://github.com/espressif/esp-matter/files/14721767/Logs.txt)
- Host Machine OS: Linux, Firmware compiled in WSL on windows
- Commissioner app and versions if present: Testharness v2.8.1-official chip-tool
- Commissioner's logs if present:

Where is the ota delay handled? On the device side or on the side of ota provider app?
If it is also handled on the device side where do i find it in the source code?

There is already an [issue](https://github.com/project-chip/matter-test-scripts/issues/135) with a similar case without any solution yet but my test provider told me they already tested another device succesfully with their testing setup.

I appreciate any sort of help or hint :)

wqx6, 2024-03-26:
`-p` is the parameter `delayedApplyActionTimeSec` of the `ApplyUpdateResponse`. And in my test, the requestor does wait 180s until it starts to apply the update.

> Software Update is executed immediately without 180s delay but it should take 180s before executing.

Does `Software Update is executed` mean that the requestor start image transfer? If yes, there is no error. The device is supposed to start image transfer immediately if the `-q` parameter is not set to `busy`.

svenehlers, 2024-04-02:
@wqx6 Thanks for your reply. I will ask my test provider if this information resolves the problem.

dhrishi, 2024-04-24:
I am closing this now. Please re-open if required, as per the feedback from the test provider.

closed 2024-04-24 by dhrishi