espressif/esp-matter #880: ESP Matter light example don't control LED

SinRuku, 2024-03-23:
I use the light example loaded into esp32_devkitj_v1 but it can only control the button and not the LED.
The picture below is what I have reconfigured in the sdkconfig file.
Thanks!

[image]

shubhamdp, 2024-03-26:
@SinRuku ESP32 devkit do not have a LED, so there may not be any visual change there. But what you can check is if you see the logs on the console correctly stating the state of an attribute.

SinRuku, 2024-03-27:
> @SinRuku ESP32 devkit do not have a LED, so there may not be any visual change there. But what you can check is if you see the logs on the console correctly stating the state of an attribute.

@shubhamdp I used an LED and wired it to the pin of the ESP32 devkit as shown. I see the logs on the console correctly stating the status of a property but the LED still does not change the state.
[image]

In the source code, I changed CONFIG_BSP_LEDS_NUM = 1 to open the BSP_LED_NUM macro code as shown below, but nothing has changed. Can you check that for me?
[image]
Thanks

shubhamdp, 2024-05-06:
Since you hardwired the LED using breadboard, you will also have to specify the gpio number that is being used for that LED, please check how that is done for esp32c3, https://github.com/espressif/esp-matter/blob/main/examples/light/sdkconfig.defaults.esp32c3#L17-L20.

Similarly, you can put such config in the `sdkconfig.defaults` and I think that should toggle LED on breadboard.

dhrishi, 2024-07-26:
@SinRuku Are you able to get this to work? Please close the issue if it works

dhrishi, 2024-10-15:
Closing this now

closed 2024-10-15 by dhrishi