espressif/esp-matter #881: ESP-IDF and ESP-Matter project build fails with HEAD-HASH-NOTFOUND (IDFGH-11631)

Tim4497, 2024-03-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-577-g6b1f40b9bf

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The project builds successfully 

### What is the actual behavior?

A lot of warnings:

```
CMake Warning at esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32 is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  esp/esp-idf/tools/cmake/build.cmake:80 (git_submodule_check)
  esp/esp-idf/tools/cmake/build.cmake:240 (__build_get_idf_git_revision)
  esp/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  esp/esp-idf/tools/cmake/project.cmake:29 (include)
  CMakeLists.txt:35 (include)

CMake Warning at esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32c2/esp32c2-bt-lib is out of
  date.  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  esp/esp-idf/tools/cmake/build.cmake:80 (git_submodule_check)
  esp/esp-idf/tools/cmake/build.cmake:240 (__build_get_idf_git_revision)
  esp/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  esp/esp-idf/tools/cmake/project.cmake:29 (include)
  CMakeLists.txt:35 (include)

CMake Warning at esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32c3_family is out of date.
  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  esp/esp-idf/tools/cmake/build.cmake:80 (git_submodule_check)
  esp/esp-idf/tools/cmake/build.cmake:240 (__build_get_idf_git_revision)
  esp/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  esp/esp-idf/tools/cmake/project.cmake:29 (include)
  CMakeLists.txt:35 (include)

CMake Warning at esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32c6/esp32c6-bt-lib is out of
  date.  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  esp/esp-idf/tools/cmake/build.cmake:80 (git_submodule_check)
  esp/esp-idf/tools/cmake/build.cmake:240 (__build_get_idf_git_revision)
  esp/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  esp/esp-idf/tools/cmake/project.cmake:29 (include)
  CMakeLists.txt:35 (include)
[... 131 lines cut ...]
install.sh

#!/bin/bash
export IDF_PATH=$PWD/esp/esp-idf
export IDF_TOOLS_PATH=$PWD/esp/idf-tools
export PATH=$PWD/esp/esp-idf/tools:$PATH

./esp/esp-idf/install.sh esp32
./esp/esp-matter/install.sh
```

```
export.sh

#!/bin/bash
export IDF_TOOLS_PATH=$PWD/esp/idf-tools
export IDF_PATH=$PWD/esp/esp-idf
export PATH=$PWD/esp/esp-idf/tools:$PATH
export PATH=$PWD/esp/esp-tools/tools:$PATH
source esp/esp-idf/export.sh
source esp/esp-matter/export.sh

```

and then `idf.py build`

shripad621git, 2024-04-12:
@Tim4497 , please make sure that you follow all the steps given below:

- If you are using particular esp-matter release , please check that you are using the recommended idf version.
 for ex: For latest esp-matter it is recommended to use idf version specified [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#development-setup).
- Please make sure that your esp-idf commit used with esp-matter is not dirty.
- In esp-idf, run git submodule update --init --recursive -f and do a install and export.
- Please follow the steps in above link to setup esp-matter and build the example.

dhrishi, 2024-04-24:
@Tim4497 Please close this if you were able to resolve your issue

dhrishi, 2024-06-20:
Closing this now

closed 2024-06-20 by dhrishi