espressif/esp-matter #882: Generating Certificate Declaration

anuarrozman, 2024-03-26:
Im generating CD using chip-cert but having problem on where was it generated. Here's my command and the output it shows

`./chip-cert gen-cd -K ../../credentials/test/certification-declaration/Chip-Test-CD-Signing-Key.pem -C ../../credentials/test/certification-declaration/Chip-Test-CD-Signing-Cert.pem gen-cd -v 0xFFF1 -p 0x8000 -t 1 -O esp_dac_fff1_8000.der -f 1 
`

Output :
`chip-cert gen-cd 0.0.0
Copyright (c) 2021-2022 Project CHIP Authors. Copyright (c) 2019 Google LLC. Copyright (c) 2013-2017 Nest Labs, Inc. All rights reserved.
`

Please help me, any insight would be appreciated.

shubhamdp, 2024-03-26:
https://docs.espressif.com/projects/esp-matter/en/latest/esp32/certification.html#certification-declaration

Please check if this would help. The output file name is provided using `-O` argument. In your case the cd should be present in the same directory where you are running the command.

anuarrozman, 2024-03-26:
Thank you @shubhamdp , Ill have a look and test it. I'll close this and will open a new one if get any problem.

anuarrozman, 2024-04-16:
hi @shubhamdp ive a question regarding the certificate-id

[image]

Im not sure where I can get it. Can you give me some insight ?

shubhamdp, 2024-04-16:
@anuarrozman  During development you can use any value, if you look [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/certification.html#certification-declaration) we have used an example value.

For production use cases, please take a look at certification process https://csa-iot.org/certification/why-certify/ and check point 8, Upon Approval, you will get a CD which you can use in your product.

anuarrozman, 2024-04-18:
Thanks @shubhamdp .