espressif/esp-matter #883: get matter sdk fail

XJGpy, 2024-03-27:
run: more than 100times, but error still!!!

git clone --depth 1 https://github.com/espressif/esp-matter.git
cd esp-matter
git submodule update --init --depth 1
cd ./connectedhomeip/connectedhomeip
./scripts/checkout_submodules.py --platform esp32 darwin --shallow
cd ../..
./install.sh

result:
TimeoutError: [Errno 60] Operation timed out

shubhamdp, 2024-04-01:
@XJGpy I hope you are following the steps mentioned in the [guide](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#getting-the-repositories).

This error is probably caused by some network connectivity issue, can you please try with the VPN? If you are familiar with docker, then you can use the esp-matter ready docker image. (https://github.com/espressif/esp-matter/tree/main/tools/docker)

dhrishi, 2024-04-30:
Closing this now. As mentioned, it looks like a network connectivity issue. Please verify the connectivity at your end.

closed 2024-04-30 by dhrishi