espressif/esp-matter #884: Matter Controller Example: Unable to start controller; Error : app_main: Failed to start Matter, err:259 in device console

vaishnavi-28, 2024-03-27:
**Describe the bug**
I built and flashed 'sdkconfig.defaults.esp32s3' example to ESP32s3 with 'Enable Matter Commissioner' option. 
Building and flashing is successful but after flashing, when S3 is connected to device console (Hterm software); Observe error:
<27>[0;31mE (2130) esp_matter_core: Node does not exist or esp_matter does not start<27>[0m<\r><\n>
<27>[0;31mE (2210) app_main: Failed to start Matter, err:259<27>[0m<\r><\n>
<\r><\n>
abort() was called at PC 0x420095c1 on core 0<\r><\n>
<\r><\n>
<\r><\n>
Backtrace: 0x40375b3e:0x3fcba6c0 0x4037fe65:0x3fcba6e0 0x40386032:0x3fcba700 0x420095c1:0x3fcba770 0x421432f7:0x3fcba790 0x403823c6:0x3fcba7c0<\r><\n>

And ESP32s3 keeps rebooting again and again. I want to connect the controller to Wifi and then commission end devices but esp32s3 (host) can not start matter (keeps rebooting, not stable)

**Environment**
- ESP-Matter Commit Id: main
- ESP-IDF Commit Id: V5.1.2
- SoC (eg: ESP32 or ESP32-C3): ESP32-S3 (host) with controller example
- Device Logs (Please attach the log file):
- Host Machine OS: ubuntu 22.04 LTS

**Any additional details**: log file attached
[ErrorwithESPMatterControllerEnable.txt](https://github.com/espressif/esp-matter/files/14774157/ErrorwithESPMatterControllerEnable.txt)

...

wqx6, 2024-03-29:
It's a known issue and we will fix ASAP. Or you can try to remove the check in `app_main.cpp` as a workaround

vaishnavi-28, 2024-04-02:
Thanks for replying. Can you please elaborate what do you mean by 'remove the check-in'? What exactly is the problem?
I can build other examples with my current esp-matter & esp-idf versions.

wqx6, 2024-04-09:
I think this issue should be fixed now in https://github.com/espressif/esp-matter/commit/e39a21e7707b47837060fec2ab8f12fd1477755f.  could you please close this issue?

dhrishi, 2024-04-10:
@vaishnavi-28 Please give it a try. Close the issue if it works.

vaishnavi-28, 2024-04-11:
Thanks for your reply. I tried with the updated repo but I some errors occurred while building the example. In any case, I can run the controller example with some logical changes in controller example and no more stuck at this problem. Thanks for your active feedback

vaishnavi-28, 2024-04-15:
I tried the commit again now I can build it successfully. I use 
Command: idf.py -D SDKCONFIG_DEFAULTS="sdkconfig.defaults.esp32s3" build 
I can commission ESP Light (MAtter-Wifi) device and send commands too. 
Thanks for the help. 

But now I want to commission thread device. I build with 'sdkconfig.defaults.otbr', build is successful. 
When in device console, I start a new dataset I observe error: Error: 83886338. I will create a different issue for this.

Sincere thanks for your help :)

closed 2024-04-15 by vaishnavi-28