espressif/esp-matter #886: Matter installation is getting using ./install.sh (This has been working two weeks back)

aggaddam, 2024-03-28:
**Describe the bug**
We have been esp-matter SDK commit "6aa7601258b10eaa0cb3845c39f41109ac0bfbaf" for a while in our products, but recently we are facing an issue when we run the ./install.sh command in esp-matter

**Environment**
- ESP-Matter Commit Id: 6aa7601258b10eaa0cb3845c39f41109ac0bfbaf (main)
- ESP-IDF Commit Id: 697e3e8eace4a7d71bdffb2f53b49a0cacdaee9b  (release/v5.1)
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):
- Host Machine OS:
      Distributor ID:	Ubuntu
      Description:	Ubuntu 22.04.4 LTS
      Release:	22.04
      Codename:	jammy
- Commissioner app and versions if present: NA
- Commissioner's logs if present: NA

**Any additional details**
...
[matter_install_fail.log](https://github.com/espressif/esp-matter/files/14787157/matter_install_fail.log)

lboue, 2024-03-28:
It's related to protobuf:

`ERROR: Cannot install -r python/gen/matter_build_venv/generated_requirements.txt (line 83), protobuf and protobuf~=3.20.1 because these package versions have conflicting dependencies.`

aggaddam, 2024-03-28:
@lboue We have been using this commit ID for last 4 months.. We are facing build issue recently, two week back. Is any dependency packages have updated ?

aggaddam, 2024-03-28:
> protobuf

[image]

lboue, 2024-03-28:
[protobuf/3.20.1](https://github.com/protocolbuffers/protobuf/releases/tag/v3.20.0) is an old version released in Mar 25, 2022. 
Maybe it should be upgraded to a newer.

aggaddam, 2024-03-28:
> [protobuf/3.20.1](https://github.com/protocolbuffers/protobuf/releases/tag/v3.20.0) is an old version released in Mar 25, 2022. Maybe it should be upgraded to a newer.

Any recommendation on updating protobuff ? Do we need to change the constraint.txt file with updated version ?

lboue, 2024-03-28:
It seems also related to [googleapis-common-protos](https://github.com/googleapis/api-common-protos):

```
pip._vendor.resolvelib.resolvers.ResolutionImpossible: [RequirementInformation(requirement=SpecifierRequirement('protobuf'), parent=None), 
RequirementInformation(requirement=SpecifierRequirement('protobuf~=3.20.1'), parent=None), 
RequirementInformation(requirement=SpecifierRequirement('protobuf!=3.20.0,!=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0.dev0,>=3.19.5'), parent=LinkCandidate('https://files.pythonhosted.org/packages/dc/a6/12a0c976140511d8bc8a16ad15793b2aef29ac927baa0786ccb7ddbb6e1c/googleapis_common_protos-1.63.0-py2.py3-none-any.whl (from https://pypi.org/simple/googleapis-common-protos/) (requires-python:>=3.7)')), 
RequirementInformation(requirement=SpecifierRequirement('protobuf'), parent=None), 
RequirementInformation(requirement=SpecifierRequirement('protobuf~=3.20.1'), parent=None), 
RequirementInformation(requirement=SpecifierRequirement('protobuf!=3.20.0,!=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0.dev0,>=3.19.5'), parent=LinkCandidate('https://files.pythonhosted.org/packages/f0/43/c9d8f75ddf08e2a0a27db243c13a700c3cc7ec615b545b697cf6f715ad92/googleapis_common_protos-1.62.0-py2.py3-none-any.whl (from https://pypi.org/simple/googleapis-common-protos/) (requires-python:>=3.7)')), 
```

lboue, 2024-03-28:
> > [protobuf/3.20.1](https://github.com/protocolbuffers/protobuf/releases/tag/v3.20.0) is an old version released in Mar 25, 2022. Maybe it should be upgraded to a newer.
> 
> Any recommendation on updating protobuff ? Do we need to change the constraint.txt file with updated version ?

You can try it while waiting for an official answer from Espressif.

aggaddam, 2024-03-28:
@lboue A Similar issue is logged in the connectedhomeip repo
https://github.com/project-chip/connectedhomeip/issues/32533

aggaddam, 2024-04-01:
EspressIf team, Any update on this issue ?

shubhamdp, 2024-04-02:
Some python dependencies broke this setup. Can you try bumping the few python module versions and give it a try?

```
cd connectedhomeip/connectedhomeip
git apply | curl https://github.com/shubhamdp/connectedhomeip/commit/b636dc0d0f7d20e249e8b5a400fdfa668254d2a9.diff
```

NOTE: This patch was curated on esp-matter commit 6aa7601258b10eaa0cb3845c39f41109ac0bfbaf and may not work as is on others, so please check the diff and make changes accordingly for other revision.

shubhamdp, 2024-04-15:
@aggaddam if this is not an issue anymore and can be solved by above comment, can you close it.

dhrishi, 2024-04-24:
Closing this now

closed 2024-04-24 by dhrishi

aggaddam, 2024-04-24:
> @aggaddam if this is not an issue anymore and can be solved by above comment, can you close it.

I am still facing the issue even after applying this patch
```
cd connectedhomeip/connectedhomeip
git apply | curl https://github.com/shubhamdp/connectedhomeip/commit/b636dc0d0f7d20e249e8b5a400fdfa668254d2a9.diff
```

Here is the issue log
[matter_install_fail.txt](https://github.com/espressif/esp-matter/files/15097096/matter_install_fail.txt)