espressif/esp-matter #887: mfg_tool.py does not generate Secure Cert partition

mostorer, 2024-03-29:
**Describe the bug**
I am in the phase to test generate a factory and secure cert partitions, following the documentation published at: https://docs.espressif.com/projects/esp-matter/en/latest/esp32c3/developing.html#factory-partition

Launched the command:
**mfg_tool.py --passcode 89674523 \
              --discriminator 2245 \
              -cd ./certification-declaration/TEST_CD_FFF1_8001.der \
              -v 0xFFF1 --vendor-name Espressif \
              -p 0x8001 --product-name Bulb \
              --hw-ver 1 --hw-ver-str DevKit**

the output messages are:
_
[2024-03-29 10:22:15,027] [   INFO] - Do not include the device attestation certificates and keys in partition binaries
[2024-03-29 10:22:15,027] [   INFO] - Number of manufacturing NVS images to generate: 1
[2024-03-29 10:22:15,036] [   INFO] - Generating Config CSV...
[2024-03-29 10:22:15,036] [   INFO] - Writing chip manifest CSV header...
[2024-03-29 10:22:15,036] [   INFO] - Appending chip master CSV row...
[2024-03-29 10:22:15,036] [   INFO] - Generated QR code: MT:-24J06PF150QJ850Y10
[2024-03-29 10:22:15,036] [   INFO] - Generated manual code: 2048-915-4736
[2024-03-29 10:22:15,045] [   INFO] - Generated onboarding data and QR Code
[2024-03-29 10:22:15,045] [   INFO] - creating out/fff1_8001/csv
[2024-03-29 10:22:15,045] [   INFO] - creating out/fff1_8001/bin

Created CSV file: ===> out/fff1_8001/csv/matter_partition-1.csv

Creating NVS binary with version: V2 - Multipage Blob Support Enabled

Created NVS binary: ===> out/fff1_8001/bin/matter_partition-1.bin

Files generated in out/fff1_8001 ...
[2024-03-29 10:22:15,046] [   INFO] - Generated output files at: out/fff1_8001/3d59a7af-9e50-4c01-ae45-4113df9cb351
_

but **bin** and **csv** subdirs are not really created.

The **out** directory is populated as follow:
**ls -alR out**
_
total 16
drwxr-xr-x   4 adminxx  staff   128 29 Mar 10:22 .
drwxr-xr-x  12 adminxx  staff   384 29 Mar 10:22 ..
-rw-r--r--@  1 adminxx  staff  6148 29 Mar 10:22 .DS_Store
drwxr-xr-x   6 adminxx  staff   192 29 Mar 10:22 fff1_8001

out/fff1_8001:
total 16
drwxr-xr-x  6 adminxx  staff  192 29 Mar 10:22 .
drwxr-xr-x  4 adminxx  staff  128 29 Mar 10:22 ..
drwxr-xr-x  6 adminxx  staff  192 29 Mar 10:22 3d59a7af-9e50-4c01-ae45-4113df9cb351
-rw-r--r--  1 adminxx  staff    9 29 Mar 10:22 cn_dacs-2024-03-29-10-22-15.csv
drwxr-xr-x  7 adminxx  staff  224 29 Mar 10:22 staging
-rw-r--r--  1 adminxx  staff  514 29 Mar 10:22 summary-2024-03-29-10-22-15.csv

out/fff1_8001/3d59a7af-9e50-4c01-ae45-4113df9cb351:
total 64
drwxr-xr-x  6 adminxx  staff    192 29 Mar 10:22 .
drwxr-xr-x  6 adminxx  staff    192 29 Mar 10:22 ..
-rw-r--r--  1 adminxx  staff     92 29 Mar 10:22 3d59a7af-9e50-4c01-ae45-4113df9cb351-onb_codes.csv
-rw-r--r--  1 adminxx  staff  24576 29 Mar 10:22 3d59a7af-9e50-4c01-ae45-4113df9cb351-partition.bin
-rw-r--r--  1 adminxx  staff    334 29 Mar 10:22 3d59a7af-9e50-4c01-ae45-4113df9cb351-qrcode.png
drwxr-xr-x  3 adminxx  staff     96 29 Mar 10:22 internal

out/fff1_8001/3d59a7af-9e50-4c01-ae45-4113df9cb351/internal:
total 8
drwxr-xr-x  3 adminxx  staff   96 29 Mar 10:22 .
drwxr-xr-x  6 adminxx  staff  192 29 Mar 10:22 ..
-rw-r--r--  1 adminxx  staff  634 29 Mar 10:22 partition.csv

out/fff1_8001/staging:
total 40
drwxr-xr-x  7 adminxx  staff  224 29 Mar 10:22 .
drwxr-xr-x  6 adminxx  staff  192 29 Mar 10:22 ..
-rw-r--r--  1 adminxx  staff  289 29 Mar 10:22 config.csv
-rw-r--r--  1 adminxx  staff  289 29 Mar 10:22 config_tmp.csv
-rw-r--r--  1 adminxx  staff  442 29 Mar 10:22 master.csv
-rw-r--r--  1 adminxx  staff  442 29 Mar 10:22 master_tmp.csv
-rw-r--r--  1 adminxx  staff  259 29 Mar 10:22 pin_disc.csv
_

**Environment**
- ESP-Matter Commit Id: [e8e6d70](https://github.com/espressif/esp-matter/commit/e8e6d70e3690db68687176737b8cafbeae869f93)
- ESP-IDF Commit Id: [be06a6f](https://github.com/espressif/esp-idf/commit/be06a6f5ffe36f9554cfc91fe2036e0fc85fea60)
- SoC (eg: ESP32 or ESP32-C3): ESP32-C3
- Device Logs (Please attach the log file):
- Host Machine OS: MacOS Sonoma 14.3.1
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

shubhamdp, 2024-03-29:
@mostorer the documentation link that you pasted is not in sync, can you please take a look at [these examples](https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool/#generate-a-factory-partition-and-store-dac-certificate-and-private-key-in-secure-cert-partition-optional-argument----dac-in-secure-cert-and---target).

Basically you need to provide `--dac-in-secure-cert` option to the script in order to generate the secure cert partition.

I am assuming you have set the correct [configurations](https://docs.espressif.com/projects/esp-matter/en/latest/esp32c3/developing.html#factory-data-providers) in firmware as well.

mostorer, 2024-03-29:
@shubhamdp you are right, that is a documentation problem. now added `--dac-in-secure-cert --target esp32c3`, bin and csv subdirs are not generated the same, but the generated directory contains both factory and secure cert partitions binary files:

_
out/fff2_8001/fbc78873-87a6-4e0f-b804-e42d3453fe9b:
total 80
drwxr-xr-x   7 adminxx  staff    224 29 Mar 11:12 .
drwxr-xr-x   7 adminxx  staff    224 29 Mar 11:13 ..
-rw-r--r--   1 adminxx  staff     92 29 Mar 11:12 fbc78873-87a6-4e0f-b804-e42d3453fe9b-onb_codes.csv
-rw-r--r--   1 adminxx  staff  24576 29 Mar 11:12 fbc78873-87a6-4e0f-b804-e42d3453fe9b-partition.bin
-rw-r--r--   1 adminxx  staff    331 29 Mar 11:12 fbc78873-87a6-4e0f-b804-e42d3453fe9b-qrcode.png
-rw-r--r--   1 adminxx  staff   8192 29 Mar 11:12 fbc78873-87a6-4e0f-b804-e42d3453fe9b_esp_secure_cert.bin
drwxr-xr-x  10 adminxx  staff    320 29 Mar 11:12 internal
_

Thank you so much.

closed 2024-03-29 by mostorer