espressif/esp-matter #889: Constants for Matter wildcards

jonsmirl, 2024-03-30:
Please add constants for subscription wildcards -- any-endpoint, any-cluster, any-attrib, etc

wqx6, 2024-04-01:
You could use 0xFFFF for any-endpoint, and 0xFFFFFFFF for any-cluster and any-attribute.

For example,
Read all the attribute in the cluster 0x6 of endpoint 1 on node 1:
```
matter esp controller read-attr 1 1 6 0xFFFFFFFF
```
Read all the attributes in all the clusters of endpoint 1 on node 1:
```
matter esp controller read-attr 1 1 0xFFFFFFFF 0xFFFFFFFF
```
Read all the attributes in all the clusters of all the endpoints on node 1:
```
matter esp controller read-attr 1 0xFFFF 0xFFFFFFFF 0xFFFFFFFF
```

jonsmirl, 2024-04-01:
I am using FFFFF... and it works.  Please add some constants in the subscribe h file to make this more readable.
const wildcardAttib = 0xFFFFFFFFFFFFFF;

wqx6, 2024-04-02:
> I am using FFFFF... and it works. Please add some constants in the subscribe h file to make this more readable. const wildcardAttib = 0xFFFFFFFFFFFFFF;

Will add the comment in the header files to declare the wildcards.

wqx6, 2024-04-09:
Added the comments in the headers. Could we close this issue?

https://github.com/espressif/esp-matter/blob/main/components/esp_matter_controller/commands/esp_matter_controller_read_command.h#L60-L61

https://github.com/espressif/esp-matter/blob/main/components/esp_matter_controller/commands/esp_matter_controller_subscribe_command.h#L67-L68

closed 2024-04-09 by jonsmirl