espressif/esp-matter #890: How do you use subscriptions?

jonsmirl, 2024-03-30:
This returns esp_err_t shouldn't it be returning cmd? And if it is returning cmd how do I override OnAttributeData()?  I need to be able to set my own code for OnAttributeData().
~~~
esp_err_t send_subscribe_attr_command(uint64_t node_id, ScopedMemoryBufferWithSize<uint16_t> &endpoint_ids,
                                      ScopedMemoryBufferWithSize<uint32_t> &cluster_ids,
                                      ScopedMemoryBufferWithSize<uint32_t> &attribute_ids, uint16_t min_interval,
                                      uint16_t max_interval, bool auto_resubscribe)
{
    if (endpoint_ids.AllocatedSize() != cluster_ids.AllocatedSize() ||
        endpoint_ids.AllocatedSize() != attribute_ids.AllocatedSize()) {
        ESP_LOGE(TAG,
                 "The endpoint_id array length should be the same as the cluster_ids array length and the "
                 "attribute_ids array length");
        return ESP_ERR_INVALID_ARG;
    }
    ScopedMemoryBufferWithSize<AttributePathParams> attr_paths;
    ScopedMemoryBufferWithSize<EventPathParams> event_paths;
    attr_paths.Alloc(endpoint_ids.AllocatedSize());
    if (!attr_paths.Get()) {
        ESP_LOGE(TAG, "Failed to alloc memory for attribute paths");
        return ESP_ERR_NO_MEM;
    }
    for (size_t i = 0; i < attr_paths.AllocatedSize(); ++i) {
        attr_paths[i] = AttributePathParams(endpoint_ids[i], cluster_ids[i], attribute_ids[i]);
    }

    subscribe_command *cmd = chip::Platform::New<subscribe_command>(
        node_id, std::move(attr_paths), std::move(event_paths), min_interval, max_interval, auto_resubscribe);
    if (!cmd) {
        ESP_LOGE(TAG, "Failed to alloc memory for subscribe_command");
        return ESP_ERR_NO_MEM;
    }
    return cmd->send_command();
}
~~~

jonsmirl, 2024-03-30:
I've just copied esp_matter_controller_subscribe_command.cpp into my app and started editing on it. It looks like that is the only way to do this.

wqx6, 2024-04-01:
The current way to use subscriptions is create a `subscribe_command` object with `attribute_data_callback` registered and call `send_command()` then.

We are planing to provide client interaction APIs(read, write, subscribe) with callback input in esp-matter component, which can be called when session is established.

jonsmirl, 2024-04-01:
The controller example in examples needs to be expanded to show the correct ways to call these APIs.

wqx6, 2024-04-02:
> The controller example in examples needs to be expanded to show the correct ways to call these APIs.

The controller example is only a command line example which supports base functionality of controller/commissioner like chip-tool, we are planing to provide a thermostat example which will call the client interaction APIs after they are added.

jonsmirl, 2024-05-28:
You don't want a subscription here, instead you want this: https://github.com/espressif/esp-matter/issues/920

closed 2024-05-28 by jonsmirl