espressif/esp-matter #892: Dynamic switching of Wi-Fi and Thread for ESP32-C6

jzwyssig, 2024-03-31:
I have an ESP32C6 running Matter over Wi-Fi. After modifying the menuconfig, I successfully compiled the firmware to include Thread support. My goal is to enable simultaneous support for both Wi-Fi and Thread, with the ability to dynamically switch between the two modes. Is there a straightforward method to accomplish this?

**Thread**
```
CONFIG_OPENTHREAD_ENABLED=y
CONFIG_ENABLE_WIFI_STATION=n
CONFIG_USE_MINIMAL_MDNS=n
```

**Wi-Fi**
```
CONFIG_OPENTHREAD_ENABLED=n
CONFIG_ENABLE_WIFI_STATION=y
CONFIG_USE_MINIMAL_MDNS=y
```

nichonaugle, 2024-04-01:
@jzwyssig I have attempted to do the following and found a few issues. Firstly, much of the supporting documentation does not advise the use of Thread and Wi-Fi support simultaneously as it will have signal integrity issues. This is a very big issue when the Thread network gets larger and impacts reliability. Secondly, the ecosystem you are selecting to commission will not allow you to choose how you commission your device and will just run with the first one it detects in the initial setup process (from my experience it always chooses Wi-Fi). Finally, the only true way to do this (in my opinion) is have the option to select the networking protocol before commissioning with some kind of switch or button, which after commissioning, would be disabled. You would have to route this to code which would setup the network configuration within the device for the respective switch position.

Hope this helps!

jzwyssig, 2024-04-01:
Thank you for the fast responses!

@nichonaugle, exactly, I am aware that running Thread and Wi-Fi simultaneously is an issue as they use the same antenna. Instead, I want to use some kind of physical button to switch between the two commissioning methods. My goal is to have everything in one firmware, where either the device boots in Thread mode or Wi-Fi mode.

@shubhamdp, thank you, the second link with the patches is exactly what I was looking for. I will check this out and get back to you.

shubhamdp, 2024-04-03:
@jzwyssig If you thing your question is answered, can you please close this and re-open if you have more info/issues with it?

jzwyssig, 2024-04-03:
@shubhamdp Sure!

closed 2024-04-03 by jzwyssig

supersunil, 2024-06-01:
Hi @shubhamdp ,
I have recreated this switching between WiFi and Thread based on the given commit id , I works as expected. 
I have tried to switch from thread to WiFi without factory-reset (connected the SSID by using `matter esp wifi connect`).
The device is in no response.

Is it possible to establish the case session for WiFi without factory-reset?

supersunil, 2024-06-12:
Hi @shubhamdp ,
I have recreated this switching between WiFi and Thread based on the given commit id , I works as expected.
I have tried to switch from thread to WiFi without factory-reset (connected the SSID by using matter esp wifi connect).
The device is in no response.

Is it possible to establish the case session for WiFi without factory-reset?