espressif/esp-matter #893: Cluster/Attribute cannot be NULL

anuarrozman, 2024-04-03:
Hi, so ive added thermostat endpoint. It run just fine but I keep getting error.

This is how i add the thermostat endpoint:
`    /*Thermostat*/
    thermostat::config_t thermostat_config;
    thermostat_config.thermostat.cooling.occupied_cooling_setpoint = 2500;
    thermostat_config.thermostat.heating.occupied_heating_setpoint = 2800;

    endpoint = thermostat::create(node, &thermostat_config, ENDPOINT_FLAG_NONE, NULL);`

Below are the error I keep seeing inside the logs:

`E (15059) esp_matter_core: Cluster cannot be NULL
I (15059) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x0000FFFC is 3 **********
I (15079) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000000 is 2877 **********
I (15099) chip[EM]: <<< [E:38099i S:59527 M:172167753] (S) Msg TX to 1:0000000070437D94 [5E3D] [UDP:[FE80::1CCB:A35B:164D:B6A1%st1]:59598] --- Type 0001:05 (IM:ReportData)
E (15069) esp_matter_core: Attribute cannot be NULL
E (15119) esp_matter_attribute: esp_matter_attr_val_type_t not handled: 0
E (15119) esp_matter_attribute: Could not allocate value buffer
E (15129) esp_matter_core: Cluster cannot be NULL
E (15129) esp_matter_core: Attribute cannot be NULL
E (15139) esp_matter_attribute: esp_matter_attr_val_type_t not handled: 0
E (15139) esp_matter_attribute: Could not allocate value buffer
`

I need help fixing this issue. Will this error affect my certification test process?

jadhavrohit924, 2024-04-03:
@anuarrozman This happens when you try to create/get attribute without providing cluster. Can you please share me the complete patch that you have created to build thermostat. It will be really helpful to understand and fix the issue.

anuarrozman, 2024-04-03:
Oh, thanks @jadhavrohit924 . Im not sure if this part of code helpful or not. Thank you for your time and consideration.

`void adt_matterInit()
{
    esp_err_t err = ESP_OK;
    esp_matter::node_t *node = node::get();
    esp_matter::endpoint_t *endpoint = NULL;

    adt_util_printf(__LINE__, __FILE_NAME__, "Free memory: %" PRIu32 " bytes", esp_get_free_heap_size());

    /* Initialize driver */
    app_driver_handle_t room_air_conditioner_handle = app_driver_room_air_conditioner_init();
    app_driver_handle_t button_handle = app_driver_button_init();
    app_reset_button_register(button_handle);

    /* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
    node::config_t node_config;
    node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

    /*Thermostat*/
    esp_matter::endpoint::thermostat::config_t thermostat_config;
    thermostat_config.thermostat.cooling.occupied_cooling_setpoint = 2500;
    thermostat_config.thermostat.heating.occupied_heating_setpoint = 2500;
    endpoint = esp_matter::endpoint::thermostat::create(node, &thermostat_config, ENDPOINT_FLAG_NONE, NULL);
    cluster_t *thermostat_cluster = esp_matter::cluster::get(endpoint, chip::app::Clusters::Thermostat::Id);
    esp_matter::cluster::thermostat::attribute::create_thermostat_running_state(thermostat_cluster, 0);
    esp_matter::cluster::thermostat::attribute::create_system_mode(thermostat_cluster, 0 , 1800, 2800);
    esp_matter::cluster::thermostat::attribute::create_max_cool_setpoint_limit(thermostat_cluster, 3000);
    esp_matter::cluster::thermostat::attribute::create_min_cool_setpoint_limit(thermostat_cluster, 1600);
    esp_matter::cluster::thermostat::attribute::create_max_heat_setpoint_limit(thermostat_cluster, 3000);
    esp_matter::cluster::thermostat::attribute::create_min_heat_setpoint_limit(thermostat_cluster, 1600);
    esp_matter::cluster::thermostat::attribute::create_local_temperature(thermostat_cluster, 0);

    /*Fan*/
    fan::config_t fan_config;
    fan_config.fan_control.percent_setting = 50;
    endpoint_t *fan_endpoint = fan::create(node, &fan_config, ENDPOINT_FLAG_NONE, NULL);

    /*Temperature Sensor*/
    temperature_sensor::config_t temperature_sensor_config;
    endpoint_t *temperature_sensor_endpoint = temperature_sensor::create(node, &temperature_sensor_config, ENDPOINT_FLAG_NONE, NULL);
    cluster_t *temperature_sensor_cluster = esp_matter::cluster::get(temperature_sensor_endpoint, chip::app::Clusters::TemperatureMeasurement::Id);
    esp_matter::cluster::temperature_measurement::attribute::create_temperature_measured_value(temperature_sensor_cluster, 0);

    /*Humidity Sensor*/
    humidity_sensor::config_t humidity_measurement_config;
    endpoint_t *humidity_sensor_endpoint = humidity_sensor::create(node, &humidity_measurement_config, ENDPOINT_FLAG_NONE, NULL);
    esp_matter::cluster_t *humidity_sensor_cluster = esp_matter::cluster::get(humidity_sensor_endpoint, chip::app::Clusters::RelativeHumidityMeasurement::Id);
    esp_matter::cluster::relative_humidity_measurement::attribute::create_relative_humidity_measured_value(humidity_sensor_cluster, 0);

    /* These node and endpoint handles can be used to create/add other endpoints and clusters. */
    if (!node || !endpoint || !fan_endpoint || !humidity_sensor_endpoint || !temperature_sensor_endpoint)
    {
        ESP_LOGE(TAG, "Matter node creation failed");
    }

    thermostat_endpoint_id = endpoint::get_id(endpoint);
    adt_util_printf(__LINE__, __FILE_NAME__, "Thermostat %d", thermostat_endpoint_id);

    fan_endpoint_id = endpoint::get_id(fan_endpoint);
    adt_util_printf(__LINE__, __FILE_NAME__, "Fan %d", fan_endpoint_id);

    temperature_sensor_endpoint_id = endpoint::get_id(temperature_sensor_endpoint);
    adt_util_printf(__LINE__, __FILE_NAME__, "Temp Sensor %d", temperature_sensor_endpoint_id);

    humidity_sensor_endpoint_id = endpoint::get_id(humidity_sensor_endpoint);
    adt_util_printf(__LINE__, __FILE_NAME__, "Humid Sensor %d", humidity_sensor_endpoint_id);

#if CHIP_DEVICE_CONFIG_ENABLE_THREAD
    /* Set OpenThread platform config */
    esp_openthread_platform_config_t config = {
        .radio_config = ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_OPENTHREAD_DEFAULT_HOST_CONFIG(),
        .port_config = ESP_OPENTHREAD_DEFAULT_PORT_CONFIG(),
    };
    set_openthread_platform_config(&config);
#endif

    /* Matter start */
    err = esp_matter::start(app_event_cb);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "Matter start failed: %d", err);
    }

    /* Starting driver with default values */
    app_driver_room_air_conditioner_set_defaults(thermostat_endpoint_id);

#if CONFIG_ENABLE_CHIP_SHELL
    esp_matter::console::diagnostics_register_commands();
    esp_matter::console::wifi_register_commands();
    esp_matter::console::init();
#endif
}`

jadhavrohit924, 2024-04-03:
@anuarrozman This code is not helpful, can you please tell me when are you seeing above error, for example when you try to read/write something and in that callback your trying to apply your driver logic. That code will be helpfull. Thanks.

anuarrozman, 2024-04-03:
Hi @jadhavrohit924 , here you go.

```
esp_err_t app_driver_attribute_update(app_driver_handle_t driver_handle, uint16_t endpoint_id, uint32_t cluster_id,
                                      uint32_t attribute_id, esp_matter_attr_val_t *val)
{
    esp_err_t err = ESP_OK;
    if (endpoint_id == thermostat_endpoint_id) {
        led_driver_handle_t handle = (led_driver_handle_t)driver_handle;
        if (cluster_id == OnOff::Id) {
            if (attribute_id == OnOff::Attributes::OnOff::Id) {
                err = app_driver_room_air_conditioner_set_power(handle, val);
            }
        }
        if (cluster_id == Thermostat::Id) {
            if (attribute_id == Thermostat::Attributes::SystemMode::Id) {
                if (val->val.u8 == 0) 
                {
                    printf("System Mode Off: %d\n", val->val.u8);
                } 
                
                else if (val->val.u8 == 1) 
                {
                    printf("System Mode Auto: %d\n", val->val.u8);
                } 
                
                else if (val->val.u8 == 3) 
                {
                    printf("System Mode Cool: %d\n", val->val.u8);
                } 
                
                else if (val->val.u8 == 4) 
                {
                    printf("System Mode Heat: %d\n", val->val.u8);
                }   
            }
        }
    }
    return err;
}
```

jadhavrohit924, 2024-04-04:
I meant when you’re trying to read attribute or get cluster you will get above errors, can you please check if you are missing something on that? If possible please tell me the steps to reproduce the error.

dhrishi, 2024-04-24:
@anuarrozman Please provide the requested details or close the issue if resolved

anuarrozman, 2024-04-30:
Hi @jadhavrohit924 @dhrishi , sorry for the late reply. First of all, I still get the error but only during device/hardware initialization(startup). Then the device and Matter side working just fine. Ill close this for now and will update any issues later on. Thank you.

closed 2024-04-30 by anuarrozman