espressif/esp-matter #896: mbedtls library version

eme-bennytang, 2024-04-09:
**Describe the question/query that you have**
I am trying to figure out which mbedtls version is the matter sdk using.

The reason I am asking is we got the SBOM generated from our project and found out we have 4 issues related to mbedtls.
**CVE-2023-52353
CVE-2024-23744**
CVE-2024-23775
CVE-2024-23170

So we wanted to clarify the confusion on
- Found mbedtls library in both esp-idf (components/mbedtls) and esp-matter (connectedhomeip/third_part/mbedtls)
  -- Which one is the matter sdk using?  (I believe the mbed library is used when checking the signatures (of the image or the certs))
  -- Moreover, what is the version of it?
- Is the mbedtls library only used for its signature checking functions? 
  -- Correct me if I am wrong, the handshaking process of the communication in Matter is done by CASE session, which is not a typical tls handshake, thus I wonder if the above **BOLDED** issues will be applied to it?

**Additional context**
esp-matter: #2f692ac
esp-idf: v5.1.2

wqx6, 2024-04-09:
> Which one is the matter sdk using? (I believe the mbed library is used when checking the signatures (of the image or the certs))

mbedtls in esp-idf is used.

> Moreover, what is the version of it?

For IDF v5.1.2, the version is 3.5.0

> Is the mbedtls library only used for its signature checking functions?

No, it will also be used for message encryption and decryption, and random bytes generation, etc. https://github.com/project-chip/connectedhomeip/blob/master/src/crypto/CHIPCryptoPALmbedTLS.cpp https://github.com/project-chip/connectedhomeip/blob/master/src/crypto/CHIPCryptoPALmbedTLSCert.cpp

wqx6, 2024-04-10:
@eme-bennytang Yes, I think these two issues will have no impact on Matter SDK.

dhrishi, 2024-04-24:
Closing as the questions are addressed

closed 2024-04-24 by dhrishi