espressif/esp-matter #898: Implementation specific answers to the security attestation

PhLuReh, 2024-04-10:
For certification approval a manufacturer has to answer security relevant questions in the [CSA_Matter_Security_Attestation_Approved.docx](https://groups.csa-iot.org/wg/members-all/document/27432)](https://groups.csa-iot.org/wg/members-all/document/27432) document. Some of them are implementation specific and should be pre-answered by espressif.

> 13.6.1.e Device shall use non-repeating initialization vectors for given session key.
what is the specific implementation? or is it a connectedhomeip specific thing?

> 13.6.3.a
is there a CVE report for the esp-matter implementation?

chshu, 2024-05-10:
@PhLuReh Here is a recommended response to 13.6.1e (if you are using esp-matter SDK without any change to the SDK source code):

13.6.1.e: Yes, the hardware RNG module is used for random number generation, which is true random, so it's non-repeating initialization vector.

Regarding 13.6.3.a: Yes, esp-matter SDK is built on top of [connectedhomeip](https://github.com/project-chip/connectedhomeip), we support the public vulnerability reporting flow.

> is there a CVE report for the esp-matter implementation?

We can't provide a public report for esp-matter implementation, since it's a product specific report, but not SDK specific. If you have any particular questions on the report, feel free to contact us via [technical-inquiries](https://www.espressif.com/en/contact-us/technical-inquiries), we can assist you on it.

PhLuReh, 2024-06-11:
I'm still struggling and hoping for further advice. Why is it such a complex process???

What I'm trying to establish: I will receive CD (from CSA) and DAC key + cert + PAI (from cert provider) in the near future, which I would be able to use in a Matter firmware, having flash encryption turned on.

What I think I have to do is:
- turn on flash encryption via `menuconfig`
- create encryption key and flash it
- generate the sec_cert partition and encrypt
- generate factory_data partition and encrypt

Up to now, each tool has a different problem, I'm struggling with:
- `configure_esp_secure_cert.py` creates `esp_secure_cert` partition, but also flashes. I need to pregenate the paritions.
- `mfg_tool.py` generates encryption key, but also generates DACs (wich will be provided).
- `generate_esp32_chip_factory_bin` with `-e` throws error: `Error. --keygen and --inputkey both are not allowed.` which were not specified.

Do I have to choose discriminator and passcode on my own or is there a tool, that does it like `mfg_tool`?

I'm quite lost, as there is no step-by-step guide on how to progrgres.

shubhamdp, 2024-06-25:
@PhLuReh Can you please close the issue if the questions are answered.

closed 2024-06-25 by PhLuReh