espressif/esp-matter #899: When running the ZAP_LIGHT example with ESP32H2, rebooting continues to occur.

joonghochoe, 2024-04-12:
After building and flashing the zap_light example with ESP32H2, rebooting continues to occur.

```
Rebooting...
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0xc (SW_CPU),boot:0x1f (SPI_FAST_FLASH_BOOT)
Saved PC:0x400031b6
SPIWP:0xee
mode:DIO, clock div:1
load:0x408460e0,len:0x17ac
load:0x4083cfd0,len:0xf04
load:0x4083efd0,len:0x2ca4
entry 0x4083cfda
I (22) boot: ESP-IDF v5.2.1 2nd stage bootloader
I (23) boot: compile time Apr 12 2024 07:57:40
I (25) boot: chip revision: v0.1
I (25) boot.esp32h2: SPI Speed      : 64MHz
I (30) boot.esp32h2: SPI Mode       : DIO
I (34) boot.esp32h2: SPI Flash Size : 4MB
I (39) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (63) boot:  1 nvs              WiFi data        01 02 00010000 0000c000
I (70) boot:  2 nvs_keys         NVS keys         01 04 0001c000 00001000
I (78) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (85) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (93) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (100) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (108) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (115) boot: End of partition table
I (120) esp_image: segment 0: paddr=00020020 vaddr=42140020 size=3b9e0h (244192) map
I (237) esp_image: segment 1: paddr=0005ba08 vaddr=40800000 size=04610h ( 17936) load
I (249) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=132ba0h (1256352) map
I (811) esp_image: segment 3: paddr=00192bc8 vaddr=40804610 size=0d174h ( 53620) load
I (841) esp_image: segment 4: paddr=0019fd44 vaddr=40811790 size=0218ch (  8588) load
I (847) esp_image: segment 5: paddr=001a1ed8 vaddr=50000000 size=00004h (     4) load
I (860) boot: Loaded app from partition at offset 0x20000
I (861) boot: Disabling RNG early entropy source...
I (872) cpu_start: Unicore app
W (882) clk: esp_perip_clk_init() has not been implemented yet
I (889) cpu_start: Pro cpu start user code
I (889) cpu_start: cpu freq: 96000000 Hz
I (890) cpu_start: Application information:
I (892) cpu_start: Project name:     zap_light
I (897) cpu_start: App version:      1.0
I (901) cpu_start: Compile time:     Apr 12 2024 09:52:52
I (907) cpu_start: ELF file SHA256:  ae956f9bb...
I (913) cpu_start: ESP-IDF:          v5.2.1
I (918) cpu_start: Min chip rev:     v0.0
I (922) cpu_start: Max chip rev:     v0.99
I (927) cpu_start: Chip rev:         v0.1
I (932) heap_init: Initializing. RAM available for dynamic allocation:
I (939) heap_init: At 40827EC0 len 000254C0 (149 KiB): RAM
I (945) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM
I (952) spi_flash: detected chip: generic
I (956) spi_flash: flash io: dio
W (960) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (972) sleep: Configure to isolate all GPIO pins in sleep state
I (977) sleep: Enable automatic switching of GPIO sleep configuration
I (985) coexist: coex firmware version: 77cd7f8
I (990) main_task: Started on CPU0
I (990) main_task: Calling app_main()
I (1000) led_driver_ws2812: Initializing light driver
I (1010) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1030) chip[DL]: NVS set: chip-counters/reboot-count = 566 (0x236)
I (1030) chip[DL]: Real time clock set to 946684800 (0100/00/01 00:00:00 UTC)
I (1030) BLE_INIT: Using main XTAL as clock source
I (1040) BLE_INIT: ble controller commit:[217f1bf]
[... 208 lines cut ...]
I (1781) led_driver_ws2812: led set r:0, g:0, b:0
I (1781) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000300's Attribute 0x00000008 is 2 **********
I (1791) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000300's Attribute 0x00004001 is 2 **********
I (1811) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000406's Attribute 0x00000001 is 0 **********
I (1821) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000406's Attribute 0x00000002 is 1 **********
I (1831) chip[DIS]: Updating services using commissioning mode 1
E (1831) chip[DIS]: Failed to remove advertised services: 3
I (1841) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1 cp=0
E (1851) chip[DIS]: Failed to advertise commissionable node: 3
E (1861) chip[DIS]: Failed to finalize service update: 3
I (1861) chip[IN]: CASE Server enabling CASE session setups
I (1871) chip[SVR]: Joining Multicast groups
I (1881) chip[SVR]: Server Listening...
E (1881) esp_matter_core: Node does not exist or esp_matter does not start
E (1891) app_main: Failed to start Matter, err:259
I (1891) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (1901) NimBLE: GAP procedure initiated: advertise;
I (1911) NimBLE: disc_mode=2
I (1911) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (1921) NimBLE:

I (1921) chip[DL]: CHIPoBLE advertising started
I (1931) app_main: Commissioning window opened
```
There is no modification in the zap_light example source, and it seems that rebooting occurs because abort() is executed at E (1891) app_main: Failed to start Matter, err:259. Is there any solution?

shubhamdp, 2024-04-15:
@joonghochoe can you please check if https://github.com/espressif/esp-matter/commit/958c7e95ed34904356d19cfa98a93d9979e1fad0 fixes this problem?

joonghochoe, 2024-04-15:
@shubhamdp Thanks for your help. When the modifications were applied, rebooting disappeared.
There is a small modification. CONFIG_ESP_MATTE_ENABLE_DATA_MODEL used in esp_matter_core.cpp should probably be modified to CONFIG_ESP_MATTER_ENABLE_DATA_MODEL.

joonghochoe, 2024-04-16:
@shubhamdp For users using the ESP data model, you will need to change CONFIG_ESP_MATTE_ENABLE_DATA_MODEL to CONFIG_ESP_MATTER_ENABLE_DATA_MODEL.

dhrishi, 2024-04-18:
Fixed with https://github.com/espressif/esp-matter/commit/ac3e08741c54373b164503245a2d6498e15bcb40

closed 2024-04-18 by dhrishi