espressif/esp-matter #900: OTA BDX only supports up to 1024 bytes of data transmission

MaplestoryAlen, 2024-04-12:
**Describe the bug**
   
   OTA can be upgraded successfully, but the data transmitted by BDX is too slow, with 1024 bytes (1KB) transmitted each time. My firmware is 4M in size, so upgrading OTA each time takes about 20 minutes
   In OTA Provider, the default constexpr uint32_t kMaxBdxBlockSize=1024. Modifying kMaxBdxBlockSize=2048 will prompt space failure
  Is there any way to improve the transmission speed of BDX, upgrading every 20 minutes, which cannot be met in practical applications

A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id:V1.2
- ESP-IDF Commit Id: 5.1.2
- SoC (eg: ESP32 or ESP32-C3):esp32s3
- Device Logs (Please attach the log file):
- Host Machine OS:Linux
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

MaplestoryAlen, 2024-04-12:
OTA-Provider log:

 (258073) chip[BDX]: Start polling for messages
I (258073) chip[EM]: <<< [E:31186r S:26661 M:256255973 (Ack:244053755)] (S) Msg TX to 1:0000000000000012 [E47E] [UDP:[FE80::4A27:E2FF:FEE7:3868%st1]:5540] --- Type 0001:09 (IM:InvokeCommandResponse)
I (258233) chip[EM]: >>> [E:31186r S:26661 M:244053756 (Ack:256255973)] (S) Msg RX from 1:0000000000000012 [E47E] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (258253) chip[EM]: >>> [E:31187r S:26661 M:244053757] (S) Msg RX from 1:0000000000000012 [E47E] --- Type 0002:04 (BDX:ReceiveInit)
I (258323) chip[EM]: <<< [E:31187r S:26661 M:256255974 (Ack:244053757)] (S) Msg TX to 1:0000000000000012 [E47E] [UDP:[FE80::4A27:E2FF:FEE7:3868%st1]:5540] --- Type 0002:05 (BDX:ReceiveAccept)
I (258543) chip[EM]: >>> [E:31187r S:26661 M:244053758 (Ack:256255974)] (S) Msg RX from 1:0000000000000012 [E47E] --- Type 0002:10 (BDX:BlockQuery)
E (258593) chip[CSL]: PacketBuffer: allocation too large.
I (258643) chip[EM]: <<< [E:31187r S:26661 M:256255975 (Ack:244053758)] (S) Msg TX to 1:0000000000000012 [E47E] [UDP:[FE80::4A27:E2FF:FEE7:3868%st1]:5540] --- Type 0000:40 (SecureChannel:StatusReport)
E (258713) chip[BDX]: InternalError
I (258713) ota-provider-app: OTA Image Transfer Failed, status:2
E (258713) chip[BDX]: onTransferFailed Callback not set
I (258953) chip[EM]: >>> [E:31187r S:26661 M:244053759 (Ack:256255975)] (S) Msg RX from 1:0000000000000012 [E47E] --- Type 0000:10 (SecureChannel:StandaloneAck)

shubhamdp, 2024-04-12:
Please check, 11.19.3.5. Transfer of OTA Software Update images

> Maximum Block Size requested by OTA Requestors over non-TCP transports SHALL be no larger than 1024 (2^10) bytes. OTA Providers SHALL support the Maximum Block Size of at least 1024 bytes in those cases.

MaplestoryAlen, 2024-04-15:
> Please check, 11.19.3.5. Transfer of OTA Software Update images
> 
> > Maximum Block Size requested by OTA Requestors over non-TCP transports SHALL be no larger than 1024 (2^10) bytes. OTA Providers SHALL support the Maximum Block Size of at least 1024 bytes in those cases.

    I tested the firmware of 4M and completed 1% data transfer in about 11 seconds. May I ask if there are any other parameters set to make data transfer faster based on MaxBdxBlockSize=1024?

closed 2024-05-30 by MaplestoryAlen