espressif/esp-matter #901: Thread device unable to get response from Matter Wifi device

FHFS, 2024-04-12:
Hello!

I am running the generic_switch example on the ESP32S3 as a Matter Wifi device.
I have a CoAP Server running on the ESP32S3 and I can send requests to and observe the server using libcoap 4.3.4 on my linux machine. So far so good.

Now I also have a nrf7002DK on my OpenThread network. Here I can send requests and observe my Laptop running a CoAP Server using libcoap.

The strange thing is I cant get responses on my requests from my nrf7002dk to the ESP32S3. I can see the request arrive on the ESP32S3 but I do not get any response on the nrf7002DK. The exact same thing happens when I try to ping from the nrf7002dk to the ESP32S3. At first I thought it had to do with the nrf7002DK, see [my post on devzone](https://devzone.nordicsemi.com/f/nordic-q-a/109931/coap-response-on-openthread-network)

[image]

I also tried the same thing at home in a different set up where my laptop runs a OTBR container with a RCP USB. Here the same problem is present, CoAP/Ping to the Laptop works but not to the ESP32S3.

[image]

This must have something to do with OpenThread because when I use Wifi on the nrf7002DK CoAP/Ping responses from the ESP32S3 arrive.

Is there a feature I have to activate to make this possible?

wqx6, 2024-04-22:
In Matter, a route hook was added to make our Wi-Fi devices be able to communicate with device in Thread network. https://github.com/project-chip/connectedhomeip/tree/master/src/platform/ESP32/route_hook 
Did you add the similar hooks to you CoAP server?

FHFS, 2024-04-22:
I just did a build with `CONFIG_ENABLE_ROUTE_HOOK=y` it works! Thanks again =D

closed 2024-04-22 by FHFS