espressif/esp-matter #903: Matter Google Home Test Suite | Thermostat Device Type

anuarrozman, 2024-04-18:
### **Matter Google Home Test Suite**
Hi, I'm currently running a Matter test for Google Home for the thermostat device type. Out of 4 tests, 3 have passed, and 1 has failed.
The failed part is the Matter device - Temperature Setting Test Suite, which will set the temperature of the Matter device to None.

### **The log from Google Home Test Suite**

```
Checking initial requirements...
    Running operation: SET attributes = %devicesAttributes[805FE354D576E8FF.00000000509D5604-0-1]
      Value {"805FE354D576E8FF.00000000509D5604-0-1":{"availableThermostatModes":["off","heat","cool"],"thermostatTemperatureUnit":"C"}} retrieved for the variable "devicesAttributes".
      Accessing property "805FE354D576E8FF.00000000509D5604-0-1". The value is {"availableThermostatModes":["off","heat","cool"],"thermostatTemperatureUnit":"C"}.
      Setting variable attributes to {"availableThermostatModes":["off","heat","cool"],"thermostatTemperatureUnit":"C"}.
    Running operation: SET tempRange = %attributes[thermostatTemperatureRange]
      Value {"availableThermostatModes":["off","heat","cool"],"thermostatTemperatureUnit":"C"} retrieved for the variable "attributes".
Check error - "Error: Invalid property access: thermostatTemperatureRange does not exist in object {\"availableThermostatModes\":[\"off\",\"heat\",\"cool\"],\"thermostatTemperatureUnit\":\"C\"}."
Test stopped.
```
I've also undergone this test using the all-device-type example (thermostat device type selected).

shubhamdp, 2024-04-18:
If you take a look [here](https://github.com/espressif/esp-matter/blob/main/examples/all_device_types_app/main/esp_matter_console_helpers.cpp#L209), all-device-type-app only implements the mandatory bits from the data model. Can you check if the test suit is trying to read the non-mandatory or any other feature based attribute? If you, please modify the data model and run test again?

anuarrozman, 2024-04-19:
Thanks, @shubhamdp. I'm currently investigating the Google Home Matter test suite. Initially, I didn't find any test suite settings like the CSA Matter certification process. The failure message I received is:

`Temperature Setting Test Suite, which will set the temperature of the Matter device to None.`

and

`Error: Invalid property access: thermostatTemperatureRange does not exist in the object.`

I'm not sure at which point Google wants to set the value to None.

According to my research, there's no `thermostatTemperatureRange` in either the Matter thermostat attribute or feature. Please correct me if I'm wrong.

jonsmirl, 2024-04-19:
Where are these Google Home test suites? I'd like to try running them.

anuarrozman, 2024-04-24:
Hi @jonsmirl . Sorry for the late reply. Ive a few screenshot of the test suite for Matter Thermostat device type that ive issue with.

[image]

VaishaliAvhale, 2024-04-30:
Hi @anuarrozman,

I've reviewed the Matter Google Home Test Suite. I used the thermostat device type from the matter all-device-type-app. The temperature setting test step failed on my end too. The reason for the failure is that the test suite is attempting to set the temperature beyond the limit. The max-cool-setpoint-limit is 3000, which corresponds to 30°C, but the temperature setpoint test suite is trying to set it to 40°C and 60°C, resulting in the failure.

VaishaliAvhale, 2024-07-12:
The Thermostat device type tests fail because the 40°C and 60°C setpoints are beyond the limit. We have notified Google about this issue. Once it is updated, we will share the update here.