espressif/esp-matter #905: Missing fixed attributes in Thermostat

jonsmirl, 2024-04-23:
Attributes 3, 4, 5, 6, 20, 21, 22, 35, 36, 38, 39 on the Thermostat cluster are all fixed. 

Does this need a mfg_tool update to set them?

jonsmirl, 2024-04-23:
Feature map support isn't right.

Attrib 10 is not getting made !LTNE
Attrib 13, 14 not getting made COOL & OCC, HEAT & OCC
so on for 15, 16, 17, 18, 19, 1a, 34, 37

jadhavrohit924, 2024-04-23:
@jonsmirl Thanks for the report! I'll take a look and update.

jadhavrohit924, 2024-04-23:
@jonsmirl We are expecting application should add HEAT, COOL feature first before adding other features

1. Attributes 13, 14 are getting created on [occupancy feature ](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_feature.cpp#L2838-L2851).
2. Attributes 15, 16, 17, 18 are optional for HEAT and COOL feature so we are not creating them by default 
3. Attribute 19 is created with AUTO feature.
4. 1a is optional
5. 34, 37 are created in setback feature.

> Attrib 10 is not getting made !LTNE

I'll add support for LTNE feature, but will not create attribute 10 by default since its conformance is optinal.

jonsmirl, 2024-04-23:
I haven't seen your solution yet for fixed attributes in code.

Maybe for fixed attributes in code when you create the cluster or turn on the feature, you have to include the fixed attributes as parameters?

jonsmirl, 2024-04-23:
I missed turning occupancy on in my code. I am just trying to turn everything on to ensure that all of the attributes appear.

jonsmirl, 2024-04-23:
Don't fixed attributes set from code need to look something like this? You have to supply the fixed attributes when the cluster is created.
~~~
cluster::thermostat::create(epThermo, &thermo_config, CLUSTER_FLAG_SERVER, cluster::thermostat::feature::cooling::get_id()
                    | cluster::thermostat::feature::setback::get_id(fixed values) | cluster::thermostat::feature::heating::get_id(fixed values)
                    | cluster::thermostat::feature::occupancy::get_id(fixed values)
                    | cluster::thermostat::feature::schedule_configuration::get_id(fixed values), fixed values);
~~~

jonsmirl, 2024-04-23:
Turning on occupancy didn't change anything, these are the attributes reported to a wildcard subscription. Don't know what FFFC is reporting so many times.
~~~
I (1682101) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000FFFC is 3 **********
I (1682102) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000FFFC is 3 **********
I (1682103) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000FFFC is 3 **********
I (1682104) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000FFFD is 6 **********
I (1682105) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000FFFC is 3 **********
I (1682106) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000000 is null **********
I (1682108) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000FFFC is 3 **********
I (1682279) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000FFFC is 3 **********
I (1682280) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000001B is 4 **********
I (1682281) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000FFFC is 3 **********
I (1682282) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000001C is 1 **********
I (1682284) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000FFFC is 3 **********
I (1682285) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000012 is 2000 **********
I (1682286) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000FFFC is 3 **********
I (1682287) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000011 is 2600 **********
~~~

jonsmirl, 2024-04-23:
LTNE feature is missing in esp_matter_feature.h

The cluster code is not creating all of the features
https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_cluster.cpp#L1541

Missing occupancy, schedule, setback, auto, ltne

jonsmirl, 2024-04-23:
I added code to force all on of the features except LTNE (missing).

Something is wrong in the CHIP layers, the feature map is reporting as 63.  Briefly looking, the CHIP cluster code makes no mention of the Occupancy, Schedule and SetBack features. So something is missing inside CHIP.
~~~
I (357060) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000FFFC is 63 **********
~~~
But when I print it from the esp-matter code it is 3f which is what I set it to.
~~~
JDS features 3f
W (989) esp_matter_core: Command 0x00000000 on cluster 0x00000201 already exists. Not creating again.
~~~
I am sending in 3F for the feature map and I see these attributes. So seven missing 7, 8, 10, 15, 16, 17, 18. Plus what to do with fixed. Plus these will all need to be audited for correct default values.
~~~
I (583302) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000000 is null **********
0x0001 Optional
I (583490) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000002 is 1 **********
0x0003 Fixed
0x0004 Fixed
0x0005 Fixed
0x0006 Fixed
0x0007 Missing (Cool)
0x0008 Missing (Heat)
0x0009 Deprecated
0x0010 Missing (!LTNE)
I (583482) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000011 is 2600 **********
I (583480) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000012 is 2000 **********
I (583492) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000013 is 2600 **********
I (583494) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000014 is 2000 **********
0x0015 Missing (Heat)
0x0016 Missing (Heat)
0x0017 Missing (Cool)
0x0018 Missing (Cool)
I (583505) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000019 is 25 **********
0x001A Optional
I (583303) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000001B is 4 **********
I (583305) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x0000001C is 1 **********
0x001D Optional
0x001E Optional
0x001F Fixed
I (583500) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000020 is 0 **********
I (583502) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000021 is 0 **********
I (583503) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000022 is 0 **********
0x0023 Optional
0x0024 Optional
0x0025 Optional
skip
0x0029 Optional
0x0030 Optional
0x0031 Optional
0x0032 Optional
skip
I (583485) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000034 is null **********
I (583486) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000035 is null **********
I (583488) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000036 is null **********
I (583495) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000037 is null **********
I (583497) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000038 is null **********
I (583498) esp_matter_attribute: ********** R : Endpoint 0x000A's Cluster 0x00000201's Attribute 0x00000039 is null **********
0x003a Optional
0x0040 Optional
0x0041 Optional
0x0042 Optional
0x0043 Optional
0x0044 Optional
0x0045 Optional
0x0046 Optional
0x0047 Optional
~~~

jonsmirl, 2024-04-23:
Another oddity, I see 11,12,13,14 on the wire, but I am only receiving reports for 11,12 in my app. So it is filtering the '& OCC' ones. I suspect that is related to the CHIP stack reporting 63 for my feature map when I have 3F set.  Something is wrong in the CHIP featuremap bits for this cluster. 63 is missing the bit for the OCC feature.

jadhavrohit924, 2024-04-24:
@jonsmirl By default we are creating MANDATORY attribute, commands, features in Esp-Matters component. Inside the feature_create we are creating MANDATORY things only. If you want anything that is optional in spec, you have to add them in the application using [attribute_creater](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_attribute.h#L394-L443) APIs.

And for the fixed attributes, every cluster_create API have config with them where you can pass your values while creating cluster as we did in [light example](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L166-L172).

> Missing occupancy, schedule, setback, auto, ltne

You can use add APIs to [create the features](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_feature.h#L477-L563). Also I have added LTNE feature its in internal review.

jonsmirl, 2024-04-24:
I see I was miss reading some of the optional flags. So will have to add of those as needed myself.

63 and 3F look to be the same.

It needs this:
~~~
diff --git a/components/esp_matter/esp_matter_cluster.cpp b/components/esp_matter/esp_matter_cluster.cpp
index 07b39bb..325a2c2 100644
--- a/components/esp_matter/esp_matter_cluster.cpp
+++ b/components/esp_matter/esp_matter_cluster.cpp
@@ -1547,6 +1547,18 @@ cluster_t *create(endpoint_t *endpoint, config_t *config, uint8_t flags, uint32_
     if (features & feature::cooling::get_id()) {
         feature::cooling::add(cluster, &(config->cooling));
     }
+    if (features & feature::setback::get_id()) {
+        feature::setback::add(cluster, &(config->setback));
+    }
+    if (features & feature::occupancy::get_id()) {
+        feature::occupancy::add(cluster, &(config->occupancy));
+    }
+    if (features & feature::schedule_configuration::get_id()) {
+        feature::schedule_configuration::add(cluster, &(config->schedule_configuration));
+    }
+    if (features & feature::auto_mode::get_id()) {
+        feature::auto_mode::add(cluster, &(config->auto_mode));
+    }
 
     return cluster;
 }
diff --git a/components/esp_matter/esp_matter_cluster.h b/components/esp_matter/esp_matter_cluster.h
index 7cab92f..7f2952c 100644
--- a/components/esp_matter/esp_matter_cluster.h
+++ b/components/esp_matter/esp_matter_cluster.h
@@ -367,6 +367,10 @@ typedef struct config {
     uint8_t system_mode;
     feature::heating::config_t heating;
     feature::cooling::config_t cooling;
+    feature::occupancy::config_t occupancy;
+    feature::setback::config_t setback;
+    feature::schedule_configuration::config_t schedule_configuration;
+    feature::auto_mode::config_t auto_mode;
     config() : cluster_revision(6), local_temperature(), control_sequence_of_operation(4), system_mode(1) {}
 } config_t;
 ~~~

jadhavrohit924, 2024-04-24:
Thank you!!! Changes will be out soon...

jonsmirl, 2024-04-24:
Next, I am trying to figure how to implement binding of remote temp sensors. This is where I need loopback binding to work since my temp sensors can be bridged onto the thermostat node. My temp sensors are BLE based, Thread temp sensors have awful battery life. The bridge code turns the BLE temp sensor into a Matter one. https://github.com/espressif/esp-matter/issues/904

jadhavrohit924, 2024-04-29:
@jonsmirl Fixed with [c47b98b](https://github.com/espressif/esp-matter/commit/c47b98bb42468273e62d3f9f0d365f68a3c37be1). Please close the issue. Lets track binding issue on #904.

closed 2024-04-29 by jonsmirl