espressif/esp-matter #906: How can I run a Nimble BLE application once BLE is no longer used by Matter?

chrelu, 2024-04-24:
Hello everyone

How can I run a Nimble BLE application once BLE is no longer used by Matter (after commissioning)?

I ask this because my current implementation doesn't work. What my application does is the following. As soon as it receives the ChipDeviceEvent "kCHIPoBLEConnectionClosed", it initializes the nimble stack and starts the BLE advertising in exactly the same way as in this example: https://github.com/michael-angerer/esp32_ble_ota/tree/main/esp/main. I have set the configuration CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING to “no” because re-initialization seems not possible after a complete deinitialization (for whatever reason). Therefore, I implemented a lightweight deinitialization, which calls nimble_port_stop and then esp_nimble_deinit (instead of nimble_port_deinit() and esp_bt_mem_release). After this lightweight deinitialization I start the Nimble BLE stack with esp_nimble_init instead of nimble_port_init. Everything can be successfully initialized in this way, but ble_gap_adv_start() returns error value 530 (0x212 = 0x200 | 0x012 = BLE_HS_EROLE). Can anyone help me? Would be great to get a working code example.

Thank you and best regards.

jonsmirl, 2024-04-24:
I used app_event_cb() to track when matter was up: kServerReady and had two IP addresses (4/6) kInterfaceIpAddressChanged. At that point I know Matter is done using BLE.  Then I started scanning....
~~~
esp_err_t StartScan()
{
    uint8_t ownAddrType;
    struct ble_gap_disc_params discParams;
    int rc;

    /* Figure out address to use while advertising. */
    rc = ble_hs_id_infer_auto(0, &ownAddrType);
    if (rc != 0)
    {
        // ChipLogError(DeviceLayer, "ble_hs_id_infer_auto failed: %d", rc);
        return ESP_ERR_INVALID_STATE;
    }

    /* Set up discovery parameters. */
    memset(&discParams, 0, sizeof(discParams));

    /* Tell the controller to filter the duplicates. */
    discParams.filter_duplicates = 0;
    /* Perform passive scanning. */
    discParams.passive = 1;
    /* Use defaults for the rest of the parameters. */
    discParams.itvl = 160;
    discParams.window = 160;
    discParams.filter_policy = 0;
    discParams.limited = 0;

    esp_ble_scan_channel_setting(ONLY_SCAN_CHANNEL_39);

    /* Start the discovery process. */
    rc = ble_gap_disc(ownAddrType, BLE_HS_FOREVER, &discParams, OnBleCentralEvent, NULL);
    if (rc != 0)
    {
        ESP_LOGE(TAG, "ble_gap_disc failed: %d", rc);
        return ESP_ERR_INVALID_STATE;
    }
    return ESP_OK;
}
~~~

chrelu, 2024-04-24:
Thank you for your quick reply.

My device is not a BLE scanner, it is an advertiser, and exactly this advertising is not working. Sorry if my explanation was confusing. Here are some code snippets for a better explanation.

```c
void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)
{
    switch (event->Type)
    {
    ...
    case chip::DeviceLayer::DeviceEventType::kCHIPoBLEConnectionClosed:
        ble_deinit();
        ble_start();
        break;
    ...
    }
}
``` 
```c
void ble_deinit(void)
{
    if (!ble_hs_is_enabled())
    {
        ESP_LOGI(TAG, "BLE already deinited");
        return;
    }

    if (nimble_port_stop() != 0)
    {
        ESP_LOGE(TAG, "nimble_port_stop() failed");
        return;
    }

   // For a correct deinitialization of BLE you should call the following two lines (same what esp_matter_core.cpp > deinit_ble_if_commissioned does) but after that I couldn't initialize it again (for whatever reason)
   // nimble_port_deinit();
   // esp_bt_mem_release(ESP_BT_MODE_BLE);

    // this is my "lightweight" way to deinitialize it
    esp_nimble_deinit();
}
``` 
```c
void ble_start(void)
{
    // Because I didn't do a "full" deinitialization I call esp_nimble_init instead of nimble_port_init(). If I run my application without using matter then I call nimble_port_init instead of esp_nimble_init. Without using matter my application works fine.
    esp_nimble_init();

    ble_hs_cfg.sync_cb = sync_cb;
    ble_hs_cfg.reset_cb = reset_cb;

    gatt_svr_init();

    ble_svc_gap_device_name_set(device_name);
    nimble_port_freertos_init(host_task);
}
``` 
```c
void sync_cb(void) {
  ble_hs_id_infer_auto(0, &addr_type);
  advertise();
}
``` 
```c
void advertise() {
  struct ble_gap_adv_params adv_params;
  struct ble_hs_adv_fields fields;
  int rc;

  memset(&fields, 0, sizeof(fields));

  fields.flags = BLE_HS_ADV_F_DISC_GEN | BLE_HS_ADV_F_BREDR_UNSUP;

  fields.tx_pwr_lvl_is_present = 1;
  fields.tx_pwr_lvl = BLE_HS_ADV_TX_PWR_LVL_AUTO;

  fields.name = (uint8_t *)device_name;
  fields.name_len = strlen(device_name);
  fields.name_is_complete = 1;

  rc = ble_gap_adv_set_fields(&fields);
  if (rc != 0) {
    ESP_LOGE(LOG_TAG_GAP, "Error setting advertisement data: rc=%d", rc);
    return;
  }

  memset(&adv_params, 0, sizeof(adv_params));
  adv_params.conn_mode = BLE_GAP_CONN_MODE_UND;
  adv_params.disc_mode = BLE_GAP_DISC_MODE_GEN;

  // HERE IT FAILS WITH RETURN VALUE 530 (0x212 = 0x200 | 0x012 = BLE_HS_EROLE)
  rc = ble_gap_adv_start(addr_type, NULL, BLE_HS_FOREVER, &adv_params,
                         gap_event_handler, NULL);
  if (rc != 0) {
    ESP_LOGE(LOG_TAG_GAP, "Error enabling advertisement data: rc=%d", rc);
    return;
  }
}
```

jonsmirl, 2024-04-24:
Why do you need to stop/start BLE?  Maybe you can set set it in menuconfig to be both central and peripheral, CHIP shouldn't care. Then when CHIP is finished with it, peripheral mode will already be running.

chrelu, 2024-04-25:
My goal is to change as little as possible in the existing esp-matter implementation. To achieve this, I wanted to start Nimble BLE with my BLE profile as soon as the esp-matter SDK tells me that BLE is no longer needed for CHIP.

As an extension, the Matter BLE profile should be supplemented with an abort service so that the user (my smartphone app) can change the BLE profile.

To illustrate this, I have drawn a simple sequence diagram. The very abstract component diagram should roughly show the dependencies and the associated BLE profile and make it clear that I only want to create/change the component "myApplication".

[image]
[image]

shubhamdp, 2024-05-06:
@chrelu If this is not an issue anymore, can you please close this?

closed 2024-05-06 by chrelu

david0215, 2024-09-05:
I have a similar idea: reusing BLE after commissioning is completed.

I enabled the SDK option CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING and I called nimble_port_init() when receive chip::DeviceLayer::DeviceEventType::kBLEDeinitialized, I get the following error:
[image],

is there any way to solve it? Thank you!

chrelu, 2024-09-06:
I had the same issue. I think the nimble port is not being deinitialized correctly. That's why I set CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING=n and only reset the ble_gatts (see my code snippets above). I hope I could help you.

david0215, 2024-09-06:
Yes, I just did what you said and I able to do the advertising now.
Thank you very much!

GaryChan-work, 2024-11-20:
Hi,

I had the same issue and could use @chrelu solution to do the advertising. However, I couldn't perform the bonding with the mobile phone as I did before. I have used the nRF Connect app on Android to connect it and perform the bonding, but the connection was dropped, and the bonding was failed. The mobile didn't prompt any message for me to confirm the passkey, and the device also didn't receive BLE_GAP_EVENT_PASSKEY_ACTION as before. 

[image]

@chrelu Do you have the similar issue?
Does anyone know how to fix this issue?

Thank you very much!