espressif/esp-matter #908: esp-idf, esp-matter, connectedhomeip versions to use

jeevithkumarnm, 2024-04-25:
I am using esp-matter landing page versioning info to setup code base to build a esp-matter light example. But I have been unsuccessful so far. Can somebody confirm if v.1.3 going latest does even work with esp-idf and connectedhomeip commit mentioned in landing page read me of esp-matter?
I even tried release/v1.2 and release/v1.1, they don't seem to work as well.
Without a release tagging on esp-matter makes it really difficult to follow codebase setup.

[image]

jeevithkumarnm, 2024-04-25:
A release tag on esp-matter repo would be really helpful to setup code-base across multiple systems. Currently, we have to pick a random commit ID.

shubhamdp, 2024-04-25:
> But I have been unsuccessful so far

Can you share more info on this? What sort of failure, at what stage? I'm assuming you are following [programming guide](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html) for setting up and building the esp-matter examples.

> I even tried latest of release/v1.2 and release/v1.1, they don't seem to work as well.
 Without a release tag on esp-matter makes it really difficult to follow codebase setup.

We only cherry-pick the critical bug fixes on the release branches so it won't have a lot more delta at least in terms of development setup.

As far as supported connectedhomeip commit it is incorrectly mentioned in the doc, and we'll fix that documentation.

shubhamdp, 2024-05-06:
@jeevithkumarnm any updates here?

jeevithkumarnm, 2024-05-07:
Hi @shubhamdp,
I was able to get code base into working sate by pulling below heads with CHIP submodule in --recursive mode.
_esp-matter at 'main' --> https://github.com/espressif/esp-matter/commit/b831d1051b9571d982aaa7854e6d839365a414ae
esp-idf at v5.2.1_

Is connectedhomeip commit updated in documentation page now?

With the above code base, I am having trouble to get light example commissioned with esp-idf mDNS instead of USE_MINIMAL_MDNS. There is a separate post for it: link[https://github.com/espressif/esp-matter/issues/913]

shubhamdp, 2024-05-07:
@jeevithkumarnm connectedhomeip is used as a submodule in esp-matter, so, doing a recursive submodule update in esp-matter is enough and no need to clone/fetch connectedhomeip separately.

> Is connectedhomeip commit updated in documentation page now?

Yes, it is updated now on the main(b84a2cc) branch.

jeevithkumarnm, 2024-05-14:
I just pulled v1.2 release from both esp-matter and CHIP per landing page, but it does not compile.
```
~/esp-matter$ git checkout release/v1.2 --recurse-submodules
~/esp-matter$ git pull --recurse-submodules
~/esp-matter/connectedhomeip/connectedhomeip$ git checkout v1.2-branch --recurse-submodules
~/esp-matter$ git pull --recurse-submodules
```

[image]

Really not so helpful...

jeevithkumarnm, 2024-05-15:
Using release branch on esp-matter made things stable. We will stick to v1.2 release for now for our development.

esp-idf : a322e6b (Tag v5.2.1)
esp-matter : c5d5835(latest branch release/v1.2)

closed 2024-05-15 by jeevithkumarnm