espressif/esp-matter #909: Failing on Building Host

osugiw, 2024-04-26:
**Describe the bug**
I'm following https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html guidelines to install the ESP_matter on my WSL2 machine. Unfortunately, I have trouble while running the **./activate.sh** in esp_matter folder. The environment has created fine, but it fails on building the Host part, as shown in attachment.

**Environment**
- ESP-Matter Commit Id: Master
- ESP-IDF Commit Id: Master
- SoC (eg: ESP32 or ESP32-C3): -
- Device Logs (Please attach the log file): Attached picture
- Host Machine OS: Windows
- Commissioner app and versions if present: -
- Commissioner's logs if present: -

**Any additional details**
...

[image]

osugiw, 2024-04-26:
**UPDATE** 
I managed to repair it and install succesfully. Instead of shallow cloning, I also clone the repository along with all submodules. Then, I also did 
`python3 -m pip install -r requirements.txt`
in esp_matter repository.
I hope you guys find it useful!

closed 2024-04-26 by osugiw

bluesafire, 2025-03-29:
Done. Made 9782 targets from 417 files in 380ms
ninja: Entering directory `/home/matter/esp-matter/connectedhomeip/connectedhomeip/out/host'
[490/558] c++ obj/src/tracing/perfetto/perfetto.perfetto_tracing.cpp.o
FAILED: obj/src/tracing/perfetto/perfetto.perfetto_tracing.cpp.o
g++ -MMD -MF obj/src/tracing/perfetto/perfetto.perfetto_tracing.cpp.o.d -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -Wno-format-nonliteral -Wno-format-nonliteral -fmacro-prefix-map=../../= -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -DCHIP_MINMDNS_USE_EPHEMERAL_UNICAST_PORT=1 -DCHIP_MINMDNS_HIGH_VERBOSITY=0 -DCHIP_MINMDNS_DEFAULT_POLICY=1 -I../../src/tracing/perfetto/include -I../../third_party/perfetto/repo/sdk -I../../src/include -I../../src -Igen/include -I../../config/standalone -I../../zzz_generated/app-common -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/include -I../../third_party/inipp/repo/inipp -c ../../src/tracing/perfetto/perfetto_tracing.cpp -o obj/src/tracing/perfetto/perfetto.perfetto_tracing.cpp.o
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
[492/558] c++ obj/src/app/util/data_model.generic-callback-stubs.cpp.o