espressif/esp-matter #910: Thermostat does not reporting to the google home that the heating is actually on

dovebman, 2024-04-27:
The temperature in the room is less that I set to the thermostat. In the code I set the Running state to 1 and the running mode to 4  but in the google home app that is not displayed. 

I'm not sure if google has implemented some kind of feedback that the heating is activated, but I assume that the app should  change the color of the thermostat tile.

What do I do wrong?

[image]
[image]

jadhavrohit924, 2024-04-28:
@dovebman Can you please elaborate more about the issue that you are facing? What are you exactly trying to do? What's the observed behaviour and what’s the expected behaviour?

dovebman, 2024-04-28:
So, the google home app is not displayed that the thermostat is actually turned on (because the temperature in the room is less than what was set). In my code, I set the running mode and running state, but in the home app nothing changed.

```
if(setpoint_value.val.i16 > comp_data.temperature)
{
	
	thermostat_state_value.val.u16 = 0b0000000000000001;
	thermostat_mode_value.val.u8 = 4;
	esp_matter::attribute::update(thermostat_endpoint_id, chip::app::Clusters::Thermostat::Id, 
									chip::app::Clusters::Thermostat::Attributes::ThermostatRunningState::Id, &thermostat_state_value);
	esp_matter::attribute::update(thermostat_endpoint_id, chip::app::Clusters::Thermostat::Id, 
									chip::app::Clusters::Thermostat::Attributes::ThermostatRunningMode::Id, &thermostat_mode_value);
}
else
{
	thermostat_state_value.val.u16 = 0b0000000000000000;
	thermostat_mode_value.val.u8 = 0;
	esp_matter::attribute::update(thermostat_endpoint_id, chip::app::Clusters::Thermostat::Id, 
									chip::app::Clusters::Thermostat::Attributes::ThermostatRunningState::Id, &thermostat_state_value);
}
```

The expected behaviour is that when the `setpoint_value.val.i16 > comp_data.temperature` is true than the google home should indicate it somehow.

jonsmirl, 2024-04-28:
Doesn't setting these via u16 and u8 mess up TLV generation which is expecting map16 and enum8?

jadhavrohit924, 2024-04-29:
@dovebman Thanks for clarification. I can see logo of heat indicating the device is on heating mode in the snapshot that you have attached. Did you see change in that logo when you change the `mode` of the thermostat. You can also check the types of mode your thermostat have by clicking on that logo. AFAIK, google home doesn’t show any UI element for the `running state`. Please check and let me know.

dovebman, 2024-04-30:
@jadhavrohit924 I changed the `featureMap` to heat only, so there is only two option (heat and off). It's sad that google home doesn't have any UI element for this. It is a very important information about a thermostat.

jadhavrohit924, 2024-04-30:
@dovebman 
>  It's sad that google home doesn't have any UI element for this
[image]

This logo represents the mode of thermostat only. The logo will change according to your `SystemModes` attribute value

dovebman, 2024-04-30:
@jadhavrohit924 Jeah I know that. I just wanted to the google home indicate somehow the `RunningState`.

jadhavrohit924, 2024-04-30:
Got it!! Sadly even Apple home don't have any UI for `RunningState` AFAIK. :)

dovebman, 2024-04-30:
Nice.. Even my old wall thermostat can indicate when the heating is on....
Anyway thanks the help! :)

closed 2024-06-20 by dhrishi