espressif/esp-matter #911: Typo in total_volatile_organic_compounds_concentration_measurement namespace in esp_matter_attribute.cpp

kibergus, 2024-04-28:
The header and other locations use plural `compounds` in the namespace name. But .cpp has the functions defined in a namespace with a singular `compound`. As a result attempting to define tvoc sensor leads to undefined symbols during linking.

```
--- a/components/esp_matter/esp_matter_attribute.cpp
+++ b/components/esp_matter/esp_matter_attribute.cpp
@@ -2754,7 +2754,7 @@ attribute_t *create_level_value(cluster_t *cluster, uint8_t value)
 } /* attribute */
 } /* radon_concentration_measurement */
 
-namespace total_volatile_organic_compound_concentration_measurement {
+namespace total_volatile_organic_compounds_concentration_measurement {
 namespace attribute {
 attribute_t *create_measured_value(cluster_t *cluster, nullable<float> value)
 {
@@ -2823,7 +2823,7 @@ attribute_t *create_level_value(cluster_t *cluster, uint8_t value)
 }
 
 } /* attribute */
-} /* total_volatile_organic_compound_concentration_measurement */
+} /* total_volatile_organic_compounds_concentration_measurement */
 
 namespace operational_state {
 namespace attribute {

```

jadhavrohit924, 2024-04-29:
@kibergus Thanks for reporting. If you can, please raise PR with the fix otherwise we will fix it soon.

kibergus, 2024-05-06:
https://github.com/espressif/esp-matter/pull/918

dhrishi, 2024-05-15:
@kibergus Thanks for the PR. We will get it merged soon

dhrishi, 2024-05-15:
Merged with https://github.com/espressif/esp-matter/commit/c9cae5a87e25363db7db4fc3fe0b2641e0932dc3

closed 2024-05-15 by dhrishi