espressif/esp-matter #912: Dynamic change of attributes and/or features

jzwyssig, 2024-04-29:
Is it feasible to dynamically alter an attribute? For instance, let's take a color temperature light that's already paired. Through an OTA update, I add the capability to adjust the color temperature range. In esp-matter, I set ColorTempPhysicalMinMireds and ColorTempPhysicalMaxMireds, which functions well for newly added lights but not for existing ones.

Another scenario: Imagine a generic switch that initially only supports short-press functionality. With an OTA update, I also enable long-press functionality. However, the switch's UI in Apple Homekit doesn't adapt accordingly to support the long-press feature. This adjustment works seamlessly when adding a new generic switch.

I believe this applies to both bridged endpoints and standard nodes with endpoints. I would like to know if there's a way to _refresh_ an existing endpoint without the need to remove and re-add it.

jadhavrohit924, 2024-04-30:
@jzwyssig You cannot alter the attributes/features after your endpoint is enabled. If you want to add new functionality to a existing light you should enable the attributes/features before matter starts.

If you want to add new attributes/features after matter starts you can always use `esp_matter::start` this will recreate the esp-matter data model

jzwyssig, 2024-04-30:
@jadhavrohit924 thank you but I think you misunderstood me. I call `esp_matter::start` after I have added the attributes or features. But the code that I added won't update the UI of an already paired fabric. 
For example:
1. Pair generic switch with short press only
2. Change programm code such that long press is added as a feature
3. Restart -> generic switch with feature long press is loaded, then `esp_matter::start` is called.

Problem: The Homekit UI does not update accordingly.

jonsmirl, 2024-05-03:
Did you change the feature map to add MSL?

jonsmirl, 2024-05-03:
~~~
3.2.7.20. ColorTempPhysicalMinMireds Attribute
The ColorTempPhysicalMinMireds attribute indicates the minimum mired value supported by the
hardware. ColorTempPhysicalMinMireds corresponds to the maximum color temperature in
kelvins supported by the hardware. ColorTempPhysicalMinMireds <= ColorTemperatureMireds.
3.2.7.21. ColorTempPhysicalMaxMireds Attribute
The ColorTempPhysicalMaxMireds attribute indicates the maximum mired value supported by the
hardware. ColorTempPhysicalMaxMireds corresponds to the minimum color temperature in
kelvins supported by the hardware. ColorTemperatureMireds <= ColorTempPhysicalMaxMireds.
~~~

Those probably should have been marked Fixed in the spec. They are hardware properties.

jadhavrohit924, 2024-07-29:
@jzwyssig Is this still an issue? If so, can you please update me?

closed 2025-03-16 by jzwyssig